iotsound/iotsound #120: Is it possible to use RaspiAudio DAC ?

boscar98, 2020-03-19:
Is it possible to use RaspiAudio DAC (https://www.raspiaudio.com/home) ? I tried with the Pimoroni pHAT DAC config because it looked similar to me but it didn't work.

chrisys, 2020-03-24:
Looking at the config it looks like it should work with `dtoverlay=hifiberry-dac` - is that the one you tried?

boscar98, 2020-03-24:
Yeah i tried with this one :/

chrisys, 2020-03-25:
Ah ok, what I'd recommend doing next, is to take a look at what the install script for this device does and make sure all those steps are carried out on your device. At first glance, many of them will be taken care of, but particularly look into the asound.conf changes that are being made.

AlexProgrammerDE, 2020-03-25:
@chrisys i just looked into the installer. If you download the installer here: script.raspiaudio.com
and have a look at line 333-338:

    if [ -e $CONFIG ] && grep -q "^dtoverlay=hifiberry-dac$" $CONFIG; then
      echo "dtoverlay already active"
    else
       echo "dtoverlay=hifiberry-dac" | sudo tee -a $CONFIG
       ASK_TO_REBOOT=true
    fi

The dtoverly should be `hifiberry-dac`.

boscar98, 2020-03-25:
Well, it's maybe the raspberry itself, he's not very new. i'll try with an other one. Thanks for your help !

closed 2020-03-25 by boscar98