iotsound/iotsound #121: [Err] Error in socket shutdown: Transport endpoint is not connected

bphermansson, 2020-03-23:
This fills my logs, one every second:

23.03.20 18:37:42 (+0100)  snapcast-client  2020-03-23 17-37-42 [Err] Error in socket shutdown: Transport endpoint is not connected

Running balenaOS 2.46.1+rev1 on a Raspberry Pi v1.

tmigone, 2020-03-27:
Hello @bphermansson can you share a bit more context? Besides this errors showing on your log, is the project working correctly?  Are you seeing this errors pop up even if you are not streaming audio?
Also, what version of the Pi 1 do you have (A, A+, B, B+)?

AlexProgrammerDE, 2020-05-11:
@bphermansson i will close your issue for now. Without more information we are not able to resolve this issue.

cgbdj, 2021-08-09:
I have this same issue and I believe it takes up a lot of CPU.

Everything works fine (bluetooth and Airplay (except for Spotifyconnect but I think it's another issue). I also do not use snap cast client (as far as I am aware. Is it possible to disable it, or fix this another way? Thanks! What information do you need?

Sungod3000, 2022-10-12:
Hi, i have the same thing now. Is there a solution so far?

cgbdj, 2022-10-12:
I have still the same error but it doesn’t affect the performance. For what I use it for.

Op EE, MMM d, jjjjj