iotsound/iotsound #127: Airplay huge audio delay

thep18, 2020-03-30:
Hi everyone,

New to the forum.
I've been using BalenaSound through Airplay on a Rasp4 for a few months and everything worked fine except a noticeable audio delay (-2000ms) that I used to manually correct every time I wanted to watch a movie or something.
I recently updated to the latest version of the OS (2.48.0+rev1) and of the Supervisor (10.8.0), now the audio delay is almost double (I'm at -3700ms).
I tried turning off all the services I'm not using (BT, Spotify) with no luck. I also tried turning off the new Snapcast and Fleet Supervisor services but when I do nothing works.

I tried to install back a version of the OS but the bigger audio delay is still here.

Any ideas?

Thank a mil for the help!

AedanBarr, 2020-03-30:
I am also having this problem. I just use Spotify until this is solved.

tmigone, 2020-03-30:
Hello, @thep18, changing balenaOS version won't do much as the added delay is most likely due to the inclusion of snapcast for multi-room feature. You can disable it and see if it helps: https://github.com/balenalabs/balena-sound/#multi-room

thep18, 2020-03-30:
That did it for me. I'm back with the -2000ms delay that I'm used to.
Thanks @tmigone 

Any idea on how to fix that now ahah?

tmigone, 2020-03-30:
Glad to know that helped!

About the 2000ms or so delay, in the case of AirPlay that is actually a design choice made by the library author. You can read more about it [here](https://github.com/mikebrady/shairport-sync/issues/526) but the gist of it is that the delay is required to avoid sync problems over the network.

With regards to multi-room delay, adding that feature requires the usage of SnapCast (or a different audio server). That unfortunately has the side effect of added streaming delay. We've decided this is an acceptable tradeoff to have multi-room, specially since you can always disable it. That being said, we are always interested in making this project better. I'm personally looking into a few enhancements that could improve this and other performance issues we've been seeing. I haven't got the time yet to try them out but will do so shortly.

I'll close this issue since I don't think we can do much about it other than the recommendations I already gave. Feel free to reopen or create new issues if required! Cheers and thanks for the feedback.

closed 2020-03-30 by tmigone

thep18, 2020-11-06:
Aaaand I'm back.
I've downgraded my Rasp4 to a Rasp3B+, and upgraded Balena to 2.60.1+rev1 with the Supervisor Version at 11.14.0.
I kept the multi-room disabled, along with the services I don't use (BT, Spotify, upnp) and I don't know what to do with the new "audio" service, but if I turn it off nothing works.
I'm at -2400ms of delay and don't know how to work my way back to -2000ms (which was a sweet spot where YouTube used to sync automatically).
Any ideas on what changed?
Thanks