iotsound/iotsound #131: Spotify Authentication failing 'BadCredentials'

BitForger, 2020-04-01:
Hey all,

So I've been enjoying balena sound so far, but I keep running into a problem with authentication with spotify. I add the credentials like directed however when ever it attempts to run from the command line it just doesn't like to authenticate. Here is what I see:
```
[2020-04-01T03:17:54Z INFO  librespot_core::session] Connecting to AP "guc3-accesspoint-b-45pt.ap.spotify.com:4070"
thread 'main' panicked at 'Authentication failed with reason: BadCredentials', /Users/admin/.cargo/registry/src/github.com-1ecc6299db9ec823/librespot-core-0.1.1/src/connection/mod.rs:121:21
stack backtrace:
   0:        0x10678cc15 - <std::sys_common::backtrace::_print::DisplayBacktrace as core::fmt::Display>::fmt::hff7732c2e44ef8b9
   1:        0x1067afedd - core::fmt::write::hd42cb3dea57bae40
   2:        0x10678675b - std::io::Write::write_fmt::ha39f6009af02b1d2
   3:        0x10678ebaa - std::panicking::default_hook::{{closure}}::h389f076017b5df43
   4:        0x10678e8aa - std::panicking::default_hook::h04b06ec20c41bf02
   5:        0x10678f1fd - std::panicking::rust_panic_with_hook::hccde7faed9a5c398
   6:        0x10678edc2 - rust_begin_unwind
   7:        0x1067c87db - std::panicking::begin_panic_fmt::hf9675b7096dbfae9
   8:        0x1064c6d9c - futures::future::chain::Chain<A,B,C>::poll::he489c615a60597d1
   9:        0x1064b4bce - <futures::future::and_then::AndThen<A,B,F> as futures::future::Future>::poll::h90990de517e44472
  10:        0x1064c025a - futures::future::chain::Chain<A,B,C>::poll::h6609e53209a1d3a6
  11:        0x1064ebadb - <futures::future::map::Map<A,F> as futures::future::Future>::poll::h1fda6213bb071071
  12:        0x10637006f - <librespot::Main as futures::future::Future>::poll::h24fa3e5be6c9e480
  13:        0x10635ddd8 - futures::task_impl::std::set::h7293cb711405eabf
  14:        0x1063577f2 - <futures::future::lazy::Lazy<F,R> as futures::future::Future>::poll::he6ac6beaba9d82d5
  15:        0x10635db68 - futures::task_impl::std::set::h03c3693e24dfad79
  16:        0x10635ce71 - std::thread::local::LocalKey<T>::with::h7049f12f959ca134
  17:        0x10635c000 - tokio_current_thread::Entered<P>::block_on::hf2cb1c34b728b1f1
  18:        0x106353b9d - tokio_timer::timer::handle::with_default::h2c01ebecc6f7920b
  19:        0x10635cf61 - std::thread::local::LocalKey<T>::with::h84f975f0c838cd9c
  20:        0x106358441 - scoped_tls::ScopedKey<T>::set::h218c98116d2aa38d
  21:        0x106374eb3 - tokio_core::reactor::Core::run::hea730c728417dad4
  22:        0x1063739df - librespot::main::h8bbb94fb146af9fa
  23:        0x106354316 - std::rt::lang_start::{{closure}}::h53ad39aa96ab83ab
  24:        0x10678ecc8 - std::panicking::try::do_call::hf2b35caddf59127c
  25:        0x1067968cb - __rust_maybe_catch_panic
  26:        0x10678f5be - std::rt::lang_start_internal::h8e563903bf99e7c1
  27:        0x106374b39 - main
```
> I know this is being run on my local but the error is exactly the same in my container. I was trying to find a fix before coming here.

So running the librespot command straight in the CLI with the variables `SPOTIFY_LOGIN` and `SPOTIFY_PASSWORD` defined in my env works just fine. It doesn't like if I try to run it in a script like the container does. Here is my test script:
```bash
SPOTIFY_CREDENTIALS="--username \"$SPOTIFY_LOGIN\" --password \"$SPOTIFY_PASSWORD\""

echo $SPOTIFY_CREDENTIALS
exec /Users/admin/.cargo/bin/librespot --name "test" $SPOTIFY_CREDENTIALS
```
I run it like so
```
export SPOTIFY_LOGIN=<login>; export SPOTIFY_PASSWORD=<password>; sh test.sh
```

This logs out the correct values for the login and password that I set, but it throws the 'BadCredentials' error that I was getting before. I'm not sure if it's me using it wrong or if this is a bug with librespot that I need to bring to them.

Any suggestions?

chrisys, 2020-04-01:
@BitForger any special characters in your username or password?

peteretep, 2020-04-01:
also having this issue. no special chars.

BitForger, 2020-04-01:
@Chrisys I tried with one that had a bunch, then changed it to one with only underscores. Same result for both.

BitForger, 2020-04-04:
So I was digging into this a bit more and I think the issue is that concatenating the options into a string and then using it in the exec line isn't turning the string into cli options. It seems the librespot args interpreter is seeing it as a single string option instead of parsing the string as multiple options.

I changed my test script to this:
```bash
SPOTIFY_CREDENTIALS="--username \"$SPOTIFY_LOGIN\" --password \"$SPOTIFY_PASSWORD\""

echo $SPOTIFY_CREDENTIALS
exec /Users/admin/.cargo/bin/librespot --name "test" -p $SPOTIFY_PASSWORD -u $SPOTIFY_LOGIN
```
In this format it works perfectly fine.

I don't know how to fix something like this as I'm not a huge bash scripter so the only thing I can think of is having two different exec lines based on if credentials are present. In the README it says that credentials are required because that's how librespot works so maybe that's alright?

Baryczka, 2020-04-04:
For me it started working finally when I remove quotes and escapes from spotify/start.sh line:23
```
- SPOTIFY_CREDENTIALS="--username \"$SPOTIFY_LOGIN\" --password \"$SPOTIFY_PASSWORD\""
+ SPOTIFY_CREDENTIALS="--username $SPOTIFY_LOGIN --password $SPOTIFY_PASSWORD"
```
Otherwise it looks as variables were not assigned properly.

BitForger, 2020-04-05:
> For me it started working finally when I remove quotes and escapes from spotify/start.sh line:23

Interesting.. so it seems that escaping that so you can have passwords with spaces breaks the functionality for people with normal passwords.

jtagcat, 2020-04-22:
> For me it started working finally when I remove quotes and escapes from spotify/start.sh line:23
>
> ```
> - SPOTIFY_CREDENTIALS="--username \"$SPOTIFY_LOGIN\" --password \"$SPOTIFY_PASSWORD\""
> + [...]

Why are there 2 double quotes at the end of the password?

jtagcat, 2020-04-22:
What env variables are you using? Any [default output ones](https://github.com/balenalabs/balena-sound#set-output-volumes)?

I found that removing them magically made it work. I'm also on a non-wireless (no-bt) Pi and I had both set to 10 (CONNECTION_NOTIFY_VOLUME might be BT-only?).

BitForger, 2020-04-26:
> What env variables are you using? Any [default output ones](https://github.com/balenalabs/balena-sound#set-output-volumes)?

I was setting the output volume to 100.

AlexProgrammerDE, 2020-06-18:
@marrobHD read the comments before telling there wasnt found a fix: https://github.com/balenalabs/balena-sound/issues/131#issuecomment-609097555

BitForger, 2020-06-20:
> Does someone found a fix?

The fix is kinda complicated.. I just made some tweaks to get it to work for me...

You can check out what I did here: https://github.com/BitForger/balena-sound

ivebeenlinuxed, 2020-09-06:
Hi All! Fix is simple: put a space in your password and it works :')

tmigone, 2020-10-28:
Should be fixed as of v3.1.11. Let me know otherwise. Thanks!

closed 2020-10-28 by tmigone