iotsound/iotsound #133: EQ support

Rzaw, 2020-04-01:
Hi! This is my first time submitting feature.
I was wondering if it could be possible to add Eq to output? 

Currently I'm using Raspberry Pi Zero (Non-W) attached to USB Lan card with external sound card (Focusrite Scarlett 2i2)

AlexProgrammerDE, 2020-04-01:
@Rzaw there is already a EQ alsa package for this: https://wiki.ubuntuusers.de/Alsamixer/#Equalizer
This should theoraticly work with it, but i don´t know how to save the alsa configuration.

jholster, 2020-05-30:
Parametric EQ support would be superb!

xvisor16, 2020-12-06:
I'll be giving this one an up to! I've already tried adding it myself but seem to get stuck on the audio routing... I've installed "caps" and "libasound2-plugin-equal" with apt and tried editing the .pa files but I can't seem to figure out where to put everything to make it work. I used this info as basis: [https://www.hifiberry.com/docs/software/guide-adding-equalization-using-alsaeq/](url) I would be amazing if this could be added! Or maybe a little help on how to get it to work with the audio routing since I can access the equaliser but it had no effect on the audio so I'm doing something wrong there...

uschwar, 2021-01-29:
I had a look at pulseaudio-equalizer, it's not really a match, as it's an equalizer with a graphical interface. Apparently it can also run headless, but it will require dbus to work.
Why is dbus disabled in the balena audio block? It doesn't state *why* it's not enabled in the docs.

xvisor16, 2021-01-29:
Maybe it is easier to just use alsa eq, since it's an easy just install the package only audio routing needs to be right and it has nice command line interface just like alsamixer we already use... -------- Oorspronkelijk bericht --------Van: uschwar <notifications@github.com>Datum: vr 29 jan. 2021 19:21Aan: balenalabs/balena-sound <balena-sound@noreply.github.com>Cc: xvisor16 <github@etict.nl>, Comment <comment@noreply.github.com>Onderwerp: Re: [balenalabs/balena-sound] EQ support (#133)
I had a look at pulseaudio-equalizer, it's not really a match, as it's an equalizer with a graphical interface. Apparently it can also run headless, but it will require dbus to work.
Why is dbus disabled in the balena audio block? It doesn't state why it's not enabled in the docs.

uschwar, 2021-01-29:
Agree, but I also have issues understanding the routing.. Maybe we just have to wait for the issue to get resolved by someone who understands the details..

ygoe, 2021-12-05:
I'm not using balena sound yet so I can't look into its config files. But I'm currently trying to get the alsa-equalizer to work with the usual plugins softvol and dmix on a Raspberry Pi (RasPi OS Lite, without desktop). And so far I haven't succeeded. That alsa-equalizer doesn't seem to cooperate with anything else. If you planned to use that, it might be a rough ride.

I was hoping that balena sound came with an equalizer, but as it seems it doesn't, I'll have to consider my options. Actually it's just an experiment for me anyway. I don't know exactly what balena OS is, compared to Raspberry Pi OS, but I have the impression that it's a little bloated. (Like docker containers for simple services. Or npm packages for trivial functions.) I don't like the idea of maintaining a full Linux system only to route some audio to several speakers in different rooms. My idea is to employ a more targeted and maintenance-free solution based on an ESP32 microcontroller or so, using just a single central audio server instead of many everywhere.

npmstart-pray, 2021-12-05:
> I'm not using balena sound yet so I can't look into its config files. But I'm currently trying to get the alsa-equalizer to work with the usual plugins softvol and dmix on a Raspberry Pi (RasPi OS [...]

I've moved to HifiBerryOS and it is slick - I gather that their DACs have some Bang & Olufsen fingers in the pi(e), so it's much more audio-focused. Balena-sound has potential but needs focused development to get easy and smooth. That said, I'll probably end up giving the balena SDR ADS-B module a try...