iotsound/iotsound #142: Bad noise RPi

polmdm, 2020-04-03:
Hi,
I’ve just flashed balenaOS + balena sound following the instructions on website. I’m currently using a RPi 3 with 32-bit software and when i connect my amplifier the only thing I ear is a bad noise and even trying to play some music (both Spotify or Bluetooth) the noise won’t stop.

To check that amplifier and cable work properly I’ve tried to connect my phone and everything works properly. It’s my fault or there’s something i can do?

AlexProgrammerDE, 2020-04-04:
@polmdm i have the same issue. What PSU are you using?

polmdm, 2020-04-04:
@AlexProgrammerDE what do you mean with PSU?

So far what I've discovered:
 - with the earpods the issue not occur at all, some lag but no background noise
 - with a different cable the noise is way less

but in both cases very low volume!

AlexProgrammerDE, 2020-04-04:
@polmdm PSU=Power Supply Unit
Not certified PSU´s can cause issues. So what are you using?

polmdm, 2020-04-04:
You mean for the Raspberry?

AlexProgrammerDE, 2020-04-04:
@polmdm yes. Which power supply gives your Pi the energy?

polmdm, 2020-04-04:
I'm using a power supply bought in a starter kit for rpi, it supply 5V and 2.5 A, the model is XS-0502500SEUD should I change it?

AlexProgrammerDE, 2020-04-04:
@polmdm i think there are differences between some PSUs. Here a example https://github.com/balenalabs/balena-sound/issues/62#issuecomment-605265537 

Can you try it with other PSU´s

polmdm, 2020-04-05:
@AlexProgrammerDE since I don't have other PSU and since I would like to develop this project on a Raspberry with Docker I think I'll go for another solution. Thanks you very much!

tmigone, 2020-07-17:
Closing, feel free to reopen.

closed 2020-07-17 by tmigone