iotsound/iotsound #144: DAC compatibility

fabio323ti, 2020-04-04:
Does someone have a list of DAC Pi compatible ?
i wish to buy "AUDIOPHONICS I-SABRE V2.1 DAC ES9023 TCXO Raspberry" 
but im not sure if it works with Balena.
Thanks

AlexProgrammerDE, 2020-04-05:
@fabio323ti there is no guarantee that it will work. But everything that works with raspbian should work with balenaOS. We have made a list of DAC hats that should work with balena: https://github.com/balenalabs/balena-sound/blob/master/DAC_configuration.md#balena_host_config_dtoverlay-values

Your DAC is not in it. But the page here: https://www.audiophonics.fr/en/dac-and-interfaces-for-raspberry-pi/audiophonics-i-sabre-v21-dac-es9023-tcxo-raspberry-pi-3-pi-2-a-b-i2s-p-11917.html
Says that the I2S driver is Hifiberry DAC. So i think that is should work with: `hifiberry-dac`. It wont work instantly. You have to set it up first: https://github.com/balenalabs/balena-sound/blob/master/DAC_configuration.md#dac-configuration
It should work then. If you report me the same i will add the DAC to the list. 👍

tmigone, 2020-07-17:
@fabio323ti let us know if you tried that DAC and if it works we can add it to the list. Cheers.

closed 2020-07-17 by tmigone

xneomac, 2020-08-10:
I have tested with an [I-SABRE AMP DAC ES9023](https://www.audiophonics.fr/fr/dac-et-interfaces-pour-raspberry-pi/audiophonics-i-sabre-amp-dac-es9023-amplificateur-class-d-2x30w-tpa3118-pour-raspberry-pi-p-11800.html) which is using the same DAC than the [I-SABRE V2.1 DAC ES9023 TCXO](https://www.audiophonics.fr/fr/dac-et-interfaces-pour-raspberry-pi/audiophonics-i-sabre-v21-dac-es9023-tcxo-raspberry-pi-3-pi-2-a-b-i2s-p-11917.html) and it is working perfectly with the `hifiberry-dac`. I Guess you can add those boards to your compatible list.