iotsound/iotsound #146: Custom Bluetooth name not working

AedanBarr, 2020-04-05:
**Describe the bug**
I have a custom name set but now it just says “balenaSound Spotify cf65”. This has only happed when you updated to 2.1.1

**Expected behavior**
I expect it to show “AEDAN’S RASPBERRY PI”

**Screenshots**
[image]
[image]

**Smartphone (please complete the following information):**
 - Device: iPhone XR 
 - OS: IOS 13.4

AlexProgrammerDE, 2020-04-05:
@AedanBarr we updated the variable. BalenaSound supports now Airplay, Bluetooth, Spotify Connect and soon UPnP. `BLUETOOTH_DEVICE_NAME` is no longer right. We updated it to: `DEVICE_NAME`.

AedanBarr, 2020-04-05:
Thanks. That explains why it was not working

closed 2020-04-05 by AlexProgrammerDE