iotsound/iotsound #149: Not pairing with Android OS

radek2s, 2020-04-07:
**Not pairing with Android OS**
I've tried to connect BalenaSound (v2.1.2) project with android device. RPI is discoverable (also with custom name) as a Bluetooth Speaker but any connection via bluetooth has not been established. (I've tried with PIN method and without).  Spotify Connect works so I'am sure that the rest of the app is working well. 

**To Reproduce**
Steps to reproduce the behavior:
1. Use android OS (Oreo 9.0 Xperia XZ1)
2. Open bluetooth settings and click on 'Connect a new device'
3. Find 'BalenaSound XXXX' or something simillar 
4. Try to connect.

_* I've noticed that someone also had the same issue:_
[Forum Topic](https://forums.balena.io/t/problem-whem-pairing-with-android-os/86834)

**Expected behavior**
Connected Android device via bluetooth

**Smartphone (please complete the following information):**
 - Device: [Sony Xperia XZ1]
 - OS: [Android 9 (Oreo) 47.2.A.11.288]

AlexProgrammerDE, 2020-04-08:
@radek2s did your device give you any error message?

peteretep, 2020-04-17:
I have what seems to be the same problem:
Sony Xperia Compact X
The bluetooth will connect and then disconnect straight away.

```
 bluetooth-audio  Setting output volume to 100%
 bluetooth-audio  Connection notify volume is 75%
 bluetooth-audio  amixer: Unable to find simple control 'PCM',0
 bluetooth-audio  
 bluetooth-audio  Restarting bluetooth service
 bluetooth-audio  Bluetooth agent registered
 bluetooth-audio  Starting bluetooth agent in Secure Simple Pairing Mode (SSPM) - No PIN code provided or invalid
 bluetooth-audio  Attempting to reconnect to previous bluetooth device: 40:33:1A:4A:1C:19
 bluetooth-audio  Device is discoverable as "living-room"
```

sirbrasscat, 2020-11-19:
I am having the same issue. Is there any updates on this?

19.11.20 08:57:21 (-0800)  bluetooth-audio  Setting output volume to 75%
19.11.20 08:57:21 (-0800)  bluetooth-audio  Connection notify volume is 75%
19.11.20 08:57:24 (-0800)  bluetooth-audio  amixer: Unable to find simple control 'PCM',0
19.11.20 08:57:24 (-0800)  bluetooth-audio  
19.11.20 08:57:24 (-0800)  bluetooth-audio  Restarting bluetooth service
19.11.20 08:57:30 (-0800)  bluetooth-audio  Bluetooth agent registered
19.11.20 08:57:30 (-0800)  bluetooth-audio  Starting bluetooth agent in Secure Simple Pairing Mode (SSPM) - No PIN code provided or invalid
19.11.20 08:57:35 (-0800)  bluetooth-audio  Device is discoverable as "balenaSound de0c"

AlexProgrammerDE, 2020-12-31:
Closing due to inactive author. Please give it a try with 3.0 we changed a lot of stuff in the backend.

closed 2020-12-31 by AlexProgrammerDE