iotsound/iotsound #155: Additional device support

rmcburney86, 2020-04-09:
I came across this project and was extremely excited until I saw it was only available for Pis.
Is there any chance you could add additional device support?  I have 3 ODroid C2s and an XU4 that I would love to run this on.
I tried deploying on the XU4, but the deployment failed on the Bluetooth drivers.
Thank you
Ryan

AlexProgrammerDE, 2020-04-09:
@rmcburney86 can you give me the logs of the deployment?

rmcburney86, 2020-04-09:
> reply@reply.github.com>
> mention@noreply.github.com>
> support (#155)

Sure thing.

This appears to be the fatal step:

Step 4/15 : RUN install_packages   alsa-
utils   bluealsa   bluez   python-dbus   python-gobject   python-
gpiozero
 ---> Running in 4ce9c37863d0
Reading package lists...
Building dependency tree...
Reading state information...
E: Unable to locate package bluealsa
E: Unable to locate package python-gpiozero

"@type": "EmailMessage",
"@type": "ViewAction",
"target": "
https://github.com/balenalabs/balena-sound/issues/155#issuecomment-611578159
",
"url": "
https://github.com/balenalabs/balena-sound/issues/155#issuecomment-611578159
",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
}
]

AlexProgrammerDE, 2020-04-09:
@rmcburney86 are you sure they support bluetooth? I cant find any information about that.

rmcburney86, 2020-04-09:
@AlexProgrammerDE 
No, they don't natively.
Do you think it would make more sense to add a USB Bluetooth adapter or run the project with only WiFi connectivity?
I think I have a USB Bluetooth\Wifi adapter in my parts bin I could try.

AlexProgrammerDE, 2020-04-09:
I think we can get bluetooth working with a dongle and some time. Bluealsa is the bridge between bluetooth and your device output. We have to build it from source and then install it. I will later work on it. But if you dont need bluetooth can we just use airplay and spotify. Do you want bluetooth?

rmcburney86, 2020-04-09:
I'd be happy to start with airplay and Spotify and work on Bluetooth down the road.

AlexProgrammerDE, 2020-04-09:
@rmcburney86 lets check if everything works without bluetooth.
You have to change some code in the compose file: https://github.com/balenalabs/balena-sound/blob/master/docker-compose.yml This file is the main of the project and says which services should be build and where they are pointed. To disable bluetooth you have to delete this part of the file:
>  bluetooth-audio:
>    build: ./bluetooth-audio
>    restart: on-failure
>    network_mode: host
>    privileged: true
>    labels:
>      io.balena.features.dbus: 1
>    volumes:
>      - bluetoothcache:/var/cache/bluetooth
>      - snapcast:/var/cache/snapcast

rmcburney86, 2020-04-10:
That worked, mostly.

The project installed on the XU4 and I can connect to it using
AirPlay.  But there is no audio output,I'm guessing because the DAC a
different. 

How would I find the settings for the onboard DAC to use the audio that
runs through the HDMI?

Thank you
Ryan

Date: Thu, 09 Apr 2020 10:22:37 -0700

https://github.com/balenalabs/balena-sound/blob/master/docker-compose.yml

https://github.com/balenalabs/balena-sound/issues/155#issuecomment-611649202
",
https://github.com/balenalabs/balena-sound/issues/155#issuecomment-611649202
",

AlexProgrammerDE, 2020-04-10:
@rmcburney86 i think this comment could help you: https://github.com/balenalabs/balena-sound/issues/79#issuecomment-583712988

It must be set in Fleet Configuration / Device Configuration

AlexProgrammerDE, 2020-04-12:
@rmcburney86 does it work?

rmcburney86, 2020-04-12:
No, I still don't have audio through the HDMI.
I have tried RESIN_HOST_CONFIG_hdmi_mode 2
With and without RESIN_HOST_CONFIG_dtparam "audio=off" 
which I found on the DAC Configuration page - https://github.com/balenalabs/balena-sound/blob/master/DAC_configuration.md
But still no luck.
Any of the other DAC settings make sense to try?

rmcburney86, 2020-04-13:
And another oddity.  
My Apple devices see the XU4 as an airplay destination, but my android doesn't.
They both can see and play through my AV receiver.

AlexProgrammerDE, 2020-04-13:
@rmcburney86 Airplay is a Apple exclusive product. You can try it with Spotify connect if you have premium. I am searching now a solution for the output.

AlexProgrammerDE, 2020-12-31:
Closing due to inactivity from author.

closed 2020-12-31 by AlexProgrammerDE