iotsound/iotsound #163: BalenaSound (balenaOS 2.48.0+rev1) not working after installing JustBoom Digi HAT on Raspberry Pi 4 #473

jonrrivera, 2020-04-14:
Hi,

I have been enjoying my BalendaSound with my new Raspberry Pi 4. But when I installed the JustBoom Digi HAT and edited the configuration it suddenly stopped working. The console is outputting some errors, but I really don’t know how to debug them…

Any help would be appreciated thanks!

[image]

[image]

AlexProgrammerDE, 2020-04-14:
@jonrrivera you used the wrong dtoverlay. Please change `justboom-dac` to `justboom-digi` and report me the result.

jonrrivera, 2020-04-14:
that was it! thank you so much!

closed 2020-04-14 by jonrrivera