iotsound/iotsound #166: DAC KeyEye

albertoxamin, 2020-04-15:
I have this [Ks0314 keyestudio ReSpeaker 2-Mic Pi HAT V1.0](https://wiki.keyestudio.com/index.php/Ks0314_keyestudio_ReSpeaker_2-Mic_Pi_HAT_V1.0)

[image]

Which configuration should I use for `BALENA_HOST_CONFIG_dtoverlay`?

AlexProgrammerDE, 2020-05-01:
@albertoxamin can you please try this value: `i2s-mmap`?

EliseoCesco, 2020-06-24:
I also have this card, I see that the 3.5mm jack input has 4 pins, can I also use headphones / earphones? Or use the microphone built into the earphones? and how eventually? THANK YOU

tmigone, 2020-06-25:
Hey @EliseoCesco, mics are not supported in balenaSound. Can I ask what would be the use case?

ruibinjiang, 2020-09-28:
Has this DAC HAT support been added yet?

shawaj, 2020-12-27:
This uses the wm8960. Don't know if it's wired up the same as the waveshare version in #264 but it might work following the same process.

shawaj, 2020-12-27:
Also, just a point to note - I believe this is a clone of the [SEEED Respeaker 2](https://wiki.seeedstudio.com/ReSpeaker_2_Mics_Pi_HAT/) so supporting one would likely support both. It is [open source hardware I believe](https://github.com/SeeedDocument/MIC_HATv1.0_for_raspberrypi) so I would imagine there are a fair few clones floating around.

There is a respeaker 4 linear and respeaker 6 circular array - both of those use the X-Power AC101 DAC - so are not the same. Just for reference.

shawaj, 2021-01-07:
I think this overlay that has just been merged in RasPi repo should now provide native support for these boards https://github.com/balenalabs/balena-sound/issues/264#issuecomment-756428272

It will take some time to filter through into balenaOS and by extension balenasound, however I will keep an eye on it.

In the meantime, you can either build the dtb yourself, or wait until it is in the raspberry firmware repo and then download it https://github.com/raspberrypi/firmware

shawaj, 2021-01-13:
Now in Raspberry Pi Firmware repo - https://github.com/raspberrypi/firmware/blob/master/boot/overlays/wm8960-soundcard.dtbo

shawaj, 2021-01-13:
Added the overlay to the PR here - https://github.com/balena-os/balena-raspberrypi/pull/585/files

Eric-AMIG, 2021-08-23:
Any updates on ReSpeaker 2 Mic Hat for Balena Sound?