iotsound/iotsound #169: rPI 4 no-audio

przemmaj, 2020-04-19:
**Describe the bug**

After clear install the newest OS version on rPI4 there is no-sound neither from BT, AirPlay, SpotifyConnect. Test on Mac and iPhone. 

**To Reproduce**
https://www.balena.io/blog/turn-your-old-speakers-or-hi-fi-into-bluetooth-receivers-using-only-a-raspberry-pi/

**Expected behavior**
Audio

**Screenshots**

**Smartphone (please complete the following information):**
 - Device: iPhone XS
 - OS: 13.3.1
- Device: Mac Book Air
- OS: Catalina10.15.3 (19D76)

**Additional context**
Not using DAC. No errors in logs.

AlexProgrammerDE, 2020-04-21:
@przemmaj are you using the 64 bit os. You are maybe facing this issue: #82

AlexProgrammerDE, 2020-04-21:
Sorry just realized the issue is about the Pi3

evb-gh, 2020-04-23:
I would like to add that I used the latest version of balena `v2.48.0+rev1 (recommended)` as well as `v2.46.1+rev3` and I have the same result as user @przemmaj, no-sound neither from BT, AirPlay, or SpotifyConnect.

AlexProgrammerDE, 2020-04-23:
@evb-gh please have a look at this: https://github.com/balenalabs/balena-sound/blob/landr-branch/docs/06-troubleshooting.md#usb-soundcard
And this issue: https://github.com/balenalabs/balena-sound/issues/9

AlexProgrammerDE, 2020-05-01:
@przemmaj i wish i could help you, but without more information is helping not possible. I will close this issue for now. Feel free to reopen if you have more information. You can try also the suggestion from this comment: https://github.com/balenalabs/balena-sound/issues/169#issuecomment-618132680

closed 2020-05-01 by AlexProgrammerDE