iotsound/iotsound #170: Airplay name does not change

Inutopia, 2020-04-20:
Running balenosound on a raspberry pi 3b+, I've changed the name using the EV (BLUETOOTH_DEVICE_NAME). This changes the name for bluetooth, but not for the airplay service as suggested in the tutorial. It remains as Balenasound Airplay XXXX. Can anyone help? Appologies if I've just done something stupid! 

Many thanks,

AlexProgrammerDE, 2020-04-20:
@Inutopia balenaSound supported in the past only Bluetooth. We used this variable set the name of the device, but since we now support Airplay and Spotify this is not correct. We changed BLUETOOTH_DEVICE_NAME to DEVICE_NAME. 
We already had this issue i will mark it as a duplicate.

Inutopia, 2020-04-20:
Many thanks for the quick reply, and sorry I've posted a duplicate! 

I've changed the variable now, and it works perfectly. It might be worth changing the tutorial, located at https://www.balena.io/blog/turn-your-old-speakers-or-hi-fi-into-bluetooth-receivers-using-only-a-raspberry-pi/ as it still references the old variable and suggests it works for all interfaces. Thanks again for the help!

AlexProgrammerDE, 2020-04-20:
@Inutopia thats good!
@chrisys could you please update the variable in the tutorial too?

tmigone, 2020-07-17:
Updated!

closed 2020-07-17 by tmigone