iotsound/iotsound #176: Please add additional input methods

JinJin-o, 2020-04-27:
First, English is not my native language
However, I will try my best with Google Translator.

First, I'm teaching students at elemaentary school.
I found the balena sound and it was very interesting and I tested it with raspberry pis at school for test.

The test was very impressive, and it was amazing that the network alone could do this.

I thought it could be good to use it right at the educational site.  In fact, the student sitting in the back seat didn't hear the sound of the TV or speaker in front, so the balena sound seemed to be able to eliminate this blind spot.

But it took a delay time.  In video viewing, this can be solved by adjusting the audio sync of the player, but it was difficult to deliver the sound of the microphone.  Also, the Bluetooth connection via the Galaxy Tab was unstable.  The iPhone's airplay was stable, but there was also a delay.

If the reason for the delay is the wireless connection between Bluetooth and Airplay.  If AUX input through a USB microphone or USB device is possible, the delay problem will be solved, so it can be a very effective alternative in the classroom.

Also, I think that there will be no delay problem in home theater composition.

Therefore, I would like to have a wired input device for efficient use.

Is there a mute button on the P.S individual device?  I think it would be more convenient if I could easily turn on / off the radio button.

AlexProgrammerDE, 2020-04-27:
@JinJin-o i didnt understand everything. But it seems like you had some delay from the multiroom. We use the software Snapcast to sync the sound over the nerwork, but it causes a delay in sound. You can deactivate it with the enviroment variable `DISABLE_MULTI_ROOM` you have to set it to `true`. Please give me your experience with the sound after that.

JinJin-o, 2020-04-28:
Ah I guess I misunderstood the function. I want to place at least two balena sound modules in the back corners of the classroom. In that case, I was asking about the delay between the video source on the front TV or the contents of the microphone spoken by the teacher and the sound from the speaker on the back.

AlexProgrammerDE, 2020-04-30:
@tmigone are there any options for synchronous audio and instant audio?

tmigone, 2020-09-21:
Hi @JinJin-o, with the recent release of balenaSound v3.0 we now have support for using USB soundcards as inputs. So you could plug in a microphone for instance directly into balenaSound. You'll need to enable the feature with the `SOUND_ENABLE_SOUNDCARD_INPUT` flag: https://sound.balenalabs.io/docs/customization#plugins

Please let me know if you have any more questions. thanks!

closed 2020-09-21 by tmigone