iotsound/iotsound #181: Allow toggling multi room flag more seamlessly

sradevski, 2020-05-01:
Feedback/request from a user:

> At times I want to play music in multiple rooms at the same time, but at times my girlfriend would like to listen to certain music in let’s say the living room as I am listening to different music in let’s say the bedroom.
 Is there a more elegant way to achieve this than going into the settings and adding or removing the DISABLE_MULTI_ROOM command every time I would like to switch between local and multiroom playback?

This will require having some simple UI I guess

AlexProgrammerDE, 2020-05-01:
This could be accomplished, by setting the master ip to `localhost`. Just needs some way to force the server.

klausagnoletti, 2020-05-03:
Why not do it more Sonos-style by adding ‘zones’, with in practice is device groups which each can be played to individually. To use this, each zone would be its own BT device or Airplay/Spotify Connect device. Dunno if this is possible but surely would be nice.

AlexProgrammerDE, 2020-05-06:
@klausagnoletti i think zones will be implemented by this PR: #153

tmigone, 2020-09-22:
Changing device mode (MULTI_ROOM; STANDALONE) is now possible via the "secret" UI at `http://<DEVICE_IP>:3000/secret`. For now this is experimental, meaning that we don't advertise it too much as it still needs polishing and moar features will be added.

Groups/zones will be addressed in the future, for that follow #188.

I'll close this issue for now.

closed 2020-09-22 by tmigone