iotsound/iotsound #182: Fleet publisher/subscriber toggling every few seconds

Aulos, 2020-05-01:
I have newly deployed setup and it's going online/offlie every few seconds:
```
02.05.20 00:36:42 (+0200)  fleet-supervisor  Fleet subscriber > service.offline Fleet publisher#a210ca9a-8787-44a0-b000-05b4ea85bca1 on 8000
02.05.20 00:36:42 (+0200)  fleet-supervisor  Fleet subscriber > service.offline Fleet subscriber#c9dbddc4-212e-4124-9685-89420d76e1ee
02.05.20 00:36:42 (+0200)  fleet-supervisor  Fleet publisher > service.offline Fleet publisher#a210ca9a-8787-44a0-b000-05b4ea85bca1 on 8000
02.05.20 00:36:42 (+0200)  fleet-supervisor  Fleet publisher > service.offline Fleet subscriber#c9dbddc4-212e-4124-9685-89420d76e1ee
02.05.20 00:37:00 (+0200)  fleet-supervisor  Fleet subscriber > service.online Fleet subscriber#c9dbddc4-212e-4124-9685-89420d76e1ee
02.05.20 00:37:00 (+0200)  fleet-supervisor  Fleet publisher > service.online Fleet subscriber#c9dbddc4-212e-4124-9685-89420d76e1ee
02.05.20 00:37:00 (+0200)  fleet-supervisor  Fleet subscriber > service.online Fleet publisher#a210ca9a-8787-44a0-b000-05b4ea85bca1 on 8000
02.05.20 00:37:00 (+0200)  fleet-supervisor  Fleet publisher > service.online Fleet publisher#a210ca9a-8787-44a0-b000-05b4ea85bca1 on 8000
02.05.20 00:37:10 (+0200)  fleet-supervisor  Fleet subscriber > service.offline Fleet subscriber#c9dbddc4-212e-4124-9685-89420d76e1ee
02.05.20 00:37:10 (+0200)  fleet-supervisor  Fleet subscriber > service.offline Fleet publisher#a210ca9a-8787-44a0-b000-05b4ea85bca1 on 8000
02.05.20 00:37:10 (+0200)  fleet-supervisor  Fleet publisher > service.offline Fleet subscriber#c9dbddc4-212e-4124-9685-89420d76e1ee
02.05.20 00:37:10 (+0200)  fleet-supervisor  Fleet publisher > service.offline Fleet publisher#a210ca9a-8787-44a0-b000-05b4ea85bca1 on 8000
02.05.20 00:37:55 (+0200)  fleet-supervisor  Fleet subscriber > service.online Fleet subscriber#c9dbddc4-212e-4124-9685-89420d76e1ee
02.05.20 00:37:55 (+0200)  fleet-supervisor  Fleet subscriber > service.online Fleet publisher#a210ca9a-8787-44a0-b000-05b4ea85bca1 on 8000
02.05.20 00:37:55 (+0200)  fleet-supervisor  Fleet publisher > service.online Fleet subscriber#c9dbddc4-212e-4124-9685-89420d76e1ee
02.05.20 00:37:55 (+0200)  fleet-supervisor  Fleet publisher > service.online Fleet publisher#a210ca9a-8787-44a0-b000-05b4ea85bca1 on 8000
02.05.20 00:38:02 (+0200)  fleet-supervisor  Fleet subscriber > service.offline Fleet subscriber#c9dbddc4-212e-4124-9685-89420d76e1ee
02.05.20 00:38:02 (+0200)  fleet-supervisor  Fleet subscriber > service.offline Fleet publisher#a210ca9a-8787-44a0-b000-05b4ea85bca1 on 8000
02.05.20 00:38:02 (+0200)  fleet-supervisor  Fleet publisher > service.offline Fleet subscriber#c9dbddc4-212e-4124-9685-89420d76e1ee
02.05.20 00:38:02 (+0200)  fleet-supervisor  Fleet publisher > service.offline Fleet publisher#a210ca9a-8787-44a0-b000-05b4ea85bca1 on 8000
02.05.20 00:38:32 (+0200)  fleet-supervisor  Fleet subscriber > service.online Fleet subscriber#c9dbddc4-212e-4124-9685-89420d76e1ee
02.05.20 00:38:32 (+0200)  fleet-supervisor  Fleet publisher > service.online Fleet subscriber#c9dbddc4-212e-4124-9685-89420d76e1ee
02.05.20 00:38:32 (+0200)  fleet-supervisor  Fleet subscriber > service.online Fleet publisher#a210ca9a-8787-44a0-b000-05b4ea85bca1 on 8000
02.05.20 00:38:32 (+0200)  fleet-supervisor  Fleet publisher > service.online Fleet publisher#a210ca9a-8787-44a0-b000-05b4ea85bca1 on 8000
02.05.20 00:38:38 (+0200)  fleet-supervisor  Fleet subscriber > service.offline Fleet subscriber#c9dbddc4-212e-4124-9685-89420d76e1ee
02.05.20 00:38:38 (+0200)  fleet-supervisor  Fleet subscriber > service.offline Fleet publisher#a210ca9a-8787-44a0-b000-05b4ea85bca1 on 8000
02.05.20 00:38:38 (+0200)  fleet-supervisor  Fleet publisher > service.offline Fleet subscriber#c9dbddc4-212e-4124-9685-89420d76e1ee
02.05.20 00:38:38 (+0200)  fleet-supervisor  Fleet publisher > service.offline Fleet publisher#a210ca9a-8787-44a0-b000-05b4ea85bca1 on 8000
```

These are logs from Raspberry 3 server instance.

AlexProgrammerDE, 2020-05-02:
@tmigone you coded the fleet supervisor. Can you help Aulos?

atulwadh1, 2020-05-04:
I also have a new setup with RPi4 and have the same issue. I see the fleet supervisor service go offline and online. The music plays seamlessly on the speaker cinnected to the RPi4 bluetooth. The other rooms cut off.

Aulos, 2020-05-05:
I "solved" my problem by "turning-off-and-on-again"ing fleet service many times. I think it may be a problem with pub/sub lib.

atulwadh1, 2020-05-05:
@Aulos Good to hear It solved your problem but my problem still remains. I have stopped and started the service multiple times but it just does not stop going offline and then online

tmigone, 2020-07-17:
@atulwadh1 did you manage to solve this? Let me know

closed 2020-07-17 by tmigone