iotsound/iotsound #185: Bluetooth audio delay on RPi4

ptruka, 2020-05-03:
**Describe the problem**
Hello, I'm using balena-sound on my RPi4 together with Hifiberry DIGI+ PRO.  The audio quality is pretty good but I'm having an delay between the player (Windows 10 Laptop) and the Audio of about 1-2
[raspberry4a_rpi4-03.05.20_21_16_32_(+0200).txt](https://github.com/balenalabs/balena-sound/files/4571429/raspberry4a_rpi4-03.05.20_21_16_32_.%2B0200.txt)
 seconds. I set the Variable DISABLE_MULTI_ROOM = 1 and rebootet the device.

**To Reproduce**
Multi-room audio is disabled

**Device information**
 - Board (hardware info): Raspberry Pi 4 (2GB) + Hifiberry DIGI+ PRO
 - Device type (as seen in balenaCloud): Raspberry Pi 4
 - balenaOS version: balenaOS 2.48.0+rev1

**Additional context**
Logfile is attached. Is there something I can do besides disabling the mutliroom? 

Best Regards

AlexProgrammerDE, 2020-05-03:
@ptruka deleting `--pcm-buffer-time=1000000` could decrease the delay:
https://github.com/balenalabs/balena-sound/blob/9238f37c31a22626c1019983bc8efe9189f220ef/bluetooth-audio/start.sh#L79

ptruka, 2020-05-04:
@AlexProgrammerDE Thanks for the quick response.
I found a lot of "start.sh" on the system (find / -name "start.sh"). What is the right way to make the change? many thanks in advance

AlexProgrammerDE, 2020-05-04:
@ptruka you need to edit the start.sh file in the bluetooth-audio directory.

ptruka, 2020-05-04:
> @ptruka deleting `--pcm-buffer-time=1000000` could decrease the delay:
> https://github.com/balenalabs/balena-sound/blob/9238f37c31a22626c1019983bc8efe9189f220ef/bluetooth-audio/start.sh#L79

@AlexProgrammerDE Thank you very much. I just pushed it with the modified start.sh.
The delay is noticeably shorter now! Videos are not lipsync, but now delay is about 200 ms.

closed 2020-05-05 by tmigone