iotsound/iotsound #187: Configuring Mono output with a hifiberry DAC+

Lootwig, 2020-05-07:
I built a mono speaker using an RPi 3B+ & hifiberry DAC+.

Following the advice from https://www.hifiberry.com/docs/software/dual-mono-setup/ and creating an asound.conf in /etc/ does not have any impact, at least on playback using spotify. Is there a way to configure the system to merge stereo channels into the right channel (or just left & right simultaneously, I don't care about what happens on L)? Thank you so much.

tmigone, 2020-09-22:
Hey @Lootwig, as of balenaSound 3.0 this is much easier to do for all audio sources. We now use a common audio container so any config applied there will apply to all audio sources. This also uses pulseaudio, so you could achieve what you want by using `module-remap-sink` as seen [here](https://wiki.archlinux.org/index.php/PulseAudio/Examples#Remap_stereo_to_mono).

You'll need to edit `core/audio/balena-sound.pa` to add the remapped sink and slightly modify the file. I think with this modifications you should get what you wanted:

```
# See https://sound.balenalabs.io/docs/architecture for details

# Create balena-sound sinks
load-module module-null-sink sink_name=balena-sound.input
load-module module-null-sink sink_name=balena-sound.output
load-module module-pipe-sink file=/var/cache/snapcast/snapfifo sink_name=snapcast format=s16le rate=44100

load-module module-remap-sink master=balena-sound.input sink_name=balena-sound.mono sink_properties="device.description='Mono'" channels=2 channel_map=mono,mono

# Route audio internally, loopback sinks depend on configuration. See start.sh for details:
# balena-sound.input: For multiroom it's routed to snapcast sink, for standalone directly wired to balena-sound.output
# balena-sound.output: Set to audio sink specified by audio block
load-module module-loopback source="balena-sound.mono.monitor" %INPUT_SINK%
load-module module-loopback source="balena-sound.output.monitor" %OUTPUT_SINK%

# Route all plugin input to the default sink
set-default-sink balena-sound.input
```
Let me know if this helps!

AlexProgrammerDE, 2020-12-30:
Closed due to 3 months of inactivity.

closed 2020-12-30 by AlexProgrammerDE

tmigone, 2021-12-20:
For newer balenaSound versions try with `balena-sound.pa`:

```
# See https://sound.balenalabs.io/docs/architecture for details

# Create balena-sound sinks
load-module module-null-sink sink_name=balena-sound.input
load-module module-null-sink sink_name=balena-sound.output
load-module module-null-sink sink_name=snapcast

load-module module-remap-sink master=balena-sound.input sink_name=balena-sound.mono sink_properties="device.description='Mono'" channels=2 channel_map=mono,mono

# Route audio internally, loopback sinks depend on configuration. See start.sh for details:
# balena-sound.input: For multiroom it's routed to snapcast sink, for standalone directly wired to balena-sound.output
# balena-sound.output: Set to audio sink specified by audio block
load-module module-loopback latency_msec=%INPUT_LATENCY% source="balena-sound.mono.monitor" %INPUT_SINK%
load-module module-loopback latency_msec=%OUTPUT_LATENCY% source="balena-sound.output.monitor" %OUTPUT_SINK%

# Route all plugin input to the default sink
set-default-sink balena-sound.input
```

gelliman, 2023-10-30:
@tmigone Thanks for putting this post together! 

I'm trying to get this to work in 2023 and I'm sadly unsuccessful. Do you have any idea why that might be the case?

With original configuration, speaker works fine, just in a single channel.

I've edited the file to be exactly the same as the above. Not sure if something has changed with newer versions of balenasound that may be causing problems. Any help would be greatly appreciated!