iotsound/iotsound #188: Configure groups that are published as a targetable wireless speaker

nebbles, 2020-05-08:
# Problem

I've been looking into multi-room and noticed BalenaSound has recently achieved this at a basic level. I can also see that #181 requests some easier control of multi-room ability and #153 is attempting to implement support for a controller.

I am dubious however of the effectiveness of roping in an extra system, and instead would like to expand configurability a bit more without the need for a managed application/frontend to change things on the fly.

# Proposal

Admittedly, this proposal hinges off the ability for BalenaSound to be able to broadcast itself as multiple speakers. If it could do so then perhaps, when configuring, we could have multiple groups established too.

Scenario:
- One Pi in the living room
- One Pi in the kitchen
- One Pi in the bedroom

Each of these are configured as normal with suitable names, each of these then appear through your streaming service of choice.

Additionally though, I could configure the living room pi to create a 'group'. It would have an ID (e.g. `group-id-1`), and a (human friendly) name (e.g. `Downstairs`). The name would be used to show a separate speaker to stream to in your service of choice, so Spotify for example would show as available speakers
- Living Room
- Kitchen
- Bedroom
- Downstairs

Other Pis on the network would also be configurable to both create a speaker, and connect to a group ID. So lets say that the Kitchen is configured to be part of `group-id-1`. So,

- When I stream to any individual speaker, it plays alone.
- When I stream to _Downstairs_, then the living room Pi gets all of the the Pis that are within the group ID associated with _Downstairs_ to play in sync.

# Alternatives

I have looked hard, but nothing seems to have the solution as described. You either don't get multiroom, or you get it but for all Pis in a Balena app, or you have to use some managed service like snap cast which means more web servers and UIs to maintain, or you give up and drop all savings on the Sonos ecosystem.

Hopefully I've been able to explain the idea well enough. I'd be interested to hear the maintainers thoughts based on their experience.

AlexProgrammerDE, 2020-05-08:
@nebbles i think this would be a big feature to implement. We need multiple UI's for creating and managing groups, new code for the fleet-supervisor and maybe android/ios apps. @tmigone when i said we need a dashboard i had this case in mind. [python-snapcast](https://github.com/happyleavesaoc/python-snapcast) and code from [snapcastr](https://github.com/xkonni/snapcastr) could help us to achieve this. Do you think we should expand the multiroom feature?

nebbles, 2020-05-08:
Generally my thought had been to avoid the UIs - the groups would be configured once at setup time through environment variables.

As I should have expected, I only realised that this posed a potential problem when it came to variable volume control per room. Then again, if you sacrifice that in favour of synchronised volume value you cut the need for UIs.

AlexProgrammerDE, 2020-05-08:
@nebbles the problem i see is that if a company want's to create a fleet of balenaSound devices no user would have access to the balena cloud dashboard. Using UI's would be more work for the team, but i think your solution won't work for tech companies.

nebbles, 2020-05-08:
That’s true, end users streaming to the speaker wouldn’t have direct configuration of the groups.

The intention with this was really a ‘one time config’ where speaker groups aren’t intended to change often.

dkjsj, 2020-06-01:
I somehow had the same challenge. Right now - I've have modified the scripts - so forceing all SnapCastClients to the same SnapCastServer - and then AirPlay/Bluetooth is enabled on each SnapCastClient. The SnapCastServer has also AirPlay enabled.

Then I can choose to stream with AirPlay to the SnapCastServer which will start muti-room stream to all Clients. 

Or I can just choose to Bluetooth/AirPlay to a dedicated room.

It's a hack - and true SnapCastServer / -Client with all services like Sonos, Bluetooth, Airplay etc would be next step :)

The SnapCastServer is also integrated with Home Assistant with can manage soudlevel etc. pr. room.

ChestersGarage, 2020-09-15:
This is doable using multiple snapcast server processes.  Each process can be a group with a name.  The fleet supervisor would need need to be updated so that it doesn't just pick a single master for all clients, but rather assign them to each group, i.e. snapcast server.  Handling multiple servers on one RPI might be too much.  But at least making each server node a possible separate group would be awesome.

s8fabord, 2021-12-21:
I'm currently setting up a Balena multiroom project at home. Just have the 3 first Raspberry 4 with Hifiberry running. All works fine so far, but I'm astonished that there doesn't seem to be any possibility yet to select the device / or group of devices you want to stream your music... 
I have 8 raspberries for 8 zones in the house. Streaming to all devices is simply not an option, I'm not alone in the house (i.e.Kids are asleep), but I want to stream to Kitchen / dining room / living room together when we have guests...
This seems to bother a few users. Is there no solution yet?