iotsound/iotsound #190: Support for Audioquest Dragonfly Red DAC

samaero, 2020-05-10:
As noted in the DACs documentation, this USB DAC isn't listed under the supported devices, so I'm adding an issue here to see if adding support would be possible. Happy to help test if needed. Thank you!

AlexProgrammerDE, 2020-05-11:
@samaero the list includes only GPIO Dac's. We define your DAC as a USB Soundcard. You can read more about setting the card up here: https://landr-balenalabs-repo-balena-sound.netlify.app/docs/troubleshooting/#no-audio

samaero, 2020-05-21:
@AlexProgrammerDE Thank you! I was able to get BalenaSound to output through the Dragonfly by simply disabling onboard audio, and everything's working fine. This DAC is capable of outputting 24bit/96KHz audio, but I couldn't find anything in the docs about setting specifying custom bit/sample rates. Do you know if this is possible? Thanks!

tmigone, 2020-07-17:
Hi, custom bitrates and samples are a bit too complex for the current state of balenaSound. We might look into that in the future after the next major version, but it's not big on the priority list.

closed 2020-07-17 by tmigone