iotsound/iotsound #193: Change sample rate per device

devdelay, 2020-05-17:
I can't seem to find where to change the sample output of each device manually. I would like to change this from 44.1kHz to at least 48kHz but I have DAC that support higher sample rates

Not sure if an environment variable is appropriate to set by device or globally.

tmigone, 2020-07-17:
Hi, this is unfortunately a bit too complex for the scope of balenaSound at the moment. We might look into this type of features in the future once we have the next major version out, but it's not a big priority.

richardpringle, 2021-01-08:
hey @tmigone, why is this too complex? 

I've managed to change the sample-rate to 192000. I had to for the repo but the edit was really simple:
```diff
-load-module module-loopback latency_msec=%INPUT_LATENCY% source="balena-sound.input.monitor" %INPUT_SINK%
-load-module module-loopback latency_msec=%OUTPUT_LATENCY% source="balena-sound.output.monitor" %OUTPUT_SINK%
+load-module module-loopback latency_msec=%INPUT_LATENCY% source="balena-sound.input.monitor" %INPUT_SINK% rate=192000 format=s16le
+load-module module-loopback latency_msec=%OUTPUT_LATENCY% source="balena-sound.output.monitor" %OUTPUT_SINK% rate=192000 format=s16le
```

tmigone, 2021-02-09:
Hmm interesting @richardpringle, I was not aware of that loopback property. It looks like setting the `rate` on the output loopback is all we need as we really don't care what the sample rate is internally as long as the output is what we want. This could be easily added with an env var. I need to do a bit of testing but sounds good.

What's the deal with the format? Did you try with one of the 24bit supported formats listed here: https://www.freedesktop.org/wiki/Software/PulseAudio/Documentation/User/SupportedAudioFormats/

richardpringle, 2021-02-12:
@tmigone, yeah I tried a couple of the 24bit formats to no avail. No matter what I set there, the format in the sink description would always just be `s16le`.