iotsound/iotsound #196: Disable bluetooth-audio service by setting a variable

densmtl, 2020-05-20:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

VincentChalnot, 2020-06-09:
I'll also be interested by this feature as the bluetooth seems to impact on spotify performances.

zaggash, 2020-06-09:
Or you can remove the Bluetooth part from the docker-compose.
Just remove :
```
bluetooth-audio:
    build: ./bluetooth-audio
    restart: on-failure
    network_mode: host
    privileged: true
    labels:
      io.balena.features.dbus: 1
    volumes:
      - bluetoothcache:/var/cache/bluetooth
      - snapcast:/var/cache/snapcast
```
and delete the folder `bluetooth-audio/`

The push a new version of your app :)

tmigone, 2020-06-15:
The correct way of removing bluetooth from the application is what @zaggash explained above. Note that you don't need to delete the `bluetooth-audio` folder, deleting the `bluetooth-audio` lines from the `docker-compose.yml` file before pushing the app will do. 

Also, note that the bluetooth container should not be interfering at all with spotify or airplay performance.

closed 2020-06-15 by tmigone