iotsound/iotsound #197: Volume Control for invidividual zones

martino-dot, 2020-05-22:
**Is your feature request related to a problem? Please describe.**
I would like when I'm playing to all my zones that I can have a volume control for each zone.

**Describe the solution you'd like**
Simple volume slider for each device.

**Describe alternatives you've considered**
I know that you can connect to an individual device and control the volume through your phone or connecting to all the devices, and doing the same, but that controls the volume for everything!

**Additional context**
Since I have devices and amps in spaces that are not easily accessible, it would makes sense for me to be able to control the volume through the dashboard.

ChestersGarage, 2020-09-15:
The snapcast client can do this.  It shows every server and client with volume sliders for each.  Immediate response and smooth.

martino-dot, 2020-09-15:
Oh didn't know this. Thanks!

closed 2020-09-15 by martino-dot