iotsound/iotsound #198: USB Sound Card

BGMine13, 2020-05-23:
Hi !

My Raspberry Pi 3 B+ internal jack doesn't work, so I used a USB Sound Card before, and it worked (I was not using balenaSound). Now that I have balenaSound installed and running, functioning etc, I need my sound to go out whit my USB Sound Card.

BGMine 13

martino-dot, 2020-05-25:
I think issue #9 has an answer to your question.

tmigone, 2020-05-26:
Hey @BGMine13, unfortunately USB sound cards are not supported yet in balenaSound.

Some users have contributed work regarding it though, you can check it and might get it working. Check out this links:
- https://sound.balenalabs.io/docs/troubleshooting/#no-audio
- https://github.com/balenalabs/balena-sound/pull/192

closed 2020-05-26 by tmigone