iotsound/iotsound #203: Push audio through usb.

thecatman11, 2020-05-27:
I am a noob at this and have a rpi zero w and it's running balena sound but I what to make the audio be be forced out of micro usb, is there a way to do this with a variable?

mcohen75, 2020-05-30:
I was able to get this to work by disabling onboard audio.

Under Device Configuration for the device in question, edit DT parameters and set audio to off instead of on. I think you could probably also just remove the audio param.

tmigone, 2020-07-17:
hi @mcohen75 are you talking about USB or HDMI?

janek, 2020-08-17:
@mcohen75's suggestion worked for me to get sound working via USB (an external sound card connected via USB) on a Zero W. Thank you!

tmigone, 2020-09-22:
USB sound cards are now supported with no configuration needed as of version 3.0.

closed 2020-09-22 by tmigone