iotsound/iotsound #204: Add DAC support for device not listed

pthorntonharvard, 2020-06-03:
Hi Good folks of this project,

I can't seem to get the DAC i recently purchased to work, and don't actually see the model listed on supported DACS either, it is a Waveshare WM8960 board.

https://www.newegg.com/p/1JW-00GU-000H8

Thanks! Loving this project, up to now building and deploying has been a breeze, and the out of the box pi 3b DAC isn't even that bad but i do notice the difference in amp and compression quality so looking forward to getting this working.

Best,
Peter

AlexProgrammerDE, 2020-06-03:
@pthorntonharvard can you try it with this value please? `wm8960` (Got it from here: https://github.com/pguyot/wm8960)

pthorntonharvard, 2020-06-03:
@AlexProgrammerDE Thanks for the link and suggestion, unfortunately still didn't work.  

Additional trouble shooting i attempted from looking at other posts: adding ",unmute_amp" to the end of the value e.g. (wm8960,unmute_amp).  

I also tried setting the SYSTEM_OUTPUT_VOLUME to 100.  

Finally, there was another dtoverlay value listed in the link you provided, and I tried dtoverlay=i2s-mmap, which didn't seem to work either.

Thanks for the assistance!

AlexProgrammerDE, 2020-06-04:
@pthorntonharvard you have set audio to false?

pthorntonharvard, 2020-06-04:
@AlexProgrammerDE yes

AlexProgrammerDE, 2020-06-04:
Ok. Have you ever tested the DAC outside of balenaOS?

pthorntonharvard, 2020-06-04:
@AlexProgrammerDE I haven't but fair point.  I have a spare pi i can test with. thanks, ill give you an update when i have one.

tmigone, 2020-07-17:
hey @pthorntonharvard let us know if you managed to get this DAC to work!

closed 2020-07-17 by tmigone