iotsound/iotsound #205: Add support for Khadas tone board DAC

gbotrel, 2020-06-03:
Using Volumio for now, without bluetooth support and no multi-room -- wanted to try with the Khadas tone board, will it work?

AlexProgrammerDE, 2020-06-03:
@gbotrel ok that looks like some high end gear. It looked like its a USB soundcard which can get powered over GPIO. Right? We made a collection of ways to configurate USB soundcards here: https://sound.balenalabs.io/docs/troubleshooting/#no-audio

tmigone, 2020-06-04:
Agree, that's a nice looking board.

I don't think it would work as the main board running balenaSound since it most likely won't run balenaOS without customising the OS. The board is not on the list of our currently supported boards (https://www.balena.io/docs/reference/hardware/devices/) and there is an open issue to add support for it (https://github.com/balena-os/resin-os-device-support/issues/77).

tmigone, 2020-07-17:
Closing this as we don't support this board upstream

closed 2020-07-17 by tmigone