iotsound/iotsound #206: Connect a bluetooth transmitter to balenasound pi3b?

Sharpey79, 2020-06-04:
Forgive me if this sounds very obvious, but am wondering if the following would be feasible...

I want to stream my vinyl records in a mutliroom setup.

Turntable is connected to an amp with a phono pre-amp and a tape out RCA 

If I connect a Bluetooth transmitter to the tape out RCA on the amp, I should be able to transmit the analogue signal over Bluetooth

My question is how would I pair the pi3b with Balenasound?  Given that there is no display to choose the device (as there is on a phone pairing)... 

Once this is connected, am I right to assume that a multiple room streaming solution could be setup with extra pi3B units.... The master unit connected to source via bluetooth, and transmitting to the other rooms?

Any help much appreciated!

Cheers,

James

tmigone, 2020-06-04:
Hi @Sharpey79, balenaSound relies on the devices that connect to it to establish the bluetooth connection (i.e: choose the appropriate balenaSound bluetooth device, enter PIN code if required, etc). This works the same way it would if you had a bluetooth speaker: the speaker creates the bluetooth connection for others to connect to.

I'm curious about this bluetooth transmitter, it should have capabilities to achieve this. Do you have a link or model/make I can check out?

Sharpey79, 2020-06-05:
Hey, thanks for your reply.  I was thinking of one of the miccus products... the one I'm linking to here works with the headphone jack..   Do you reckon it would work?

https://cdn.shopify.com/s/files/1/0019/0486/9449/files/Swivel_Jack_User_Manual-120118.pdf?93

Thanks for taking the time. I do appreciate it!

tmigone, 2020-06-05:
Hi, I'm checking the manual for the Swivel and it doesn't appear to allow headless connections, you need the other device to initiate the connection. This unfortunately wont work with balenaSound as it will also expect the other device to initiate the connection. We could theoretically add a UI sort of interface to allow this but it's quite outside the scope of this project unfortunately.

As an alternative, you could get an RCA to 3.5mm cable and send the audio from the preamp to the Pi, right? Would that work for you?

AlexProgrammerDE, 2020-06-05:
@tmigone wouldnt bluetoothctl work? He Sharpey access to the terminal.

tmigone, 2020-06-05:
Yes I think with bluetoothctl it could be done. But it would require user input for sure (to run the commands, identify the target connection, and pair with it, basically everything you do with your phone but on the device). 

Unless I'm missing something I don't see how we could achieve this to be done automatically and in a generic way. If you have any ideas we can try!

AlexProgrammerDE, 2020-06-05:
@tmigone didnt you implement auto reconnection?

tmigone, 2020-06-05:
Yep, but I don't think it helps here. My main concern is how do you tell balenaSound which bluetooth it should connect to initially? 

Usually you have the device (smartphone) telling balenaSound that it wants to connect via bluetooth. balenaSound only accepts the connection. 

Here the Swivel thing can't tell balenaSound that it wants to connect because it has no physical interface to do so. So balenaSound could theoretically initiate the connection, but there is two questions:
1. How does sound know it has to scan for bluetooth networks instead of expecting an incoming connection?
2. How does sound know which bluetooth it has to connect to?

I guess you could hack something with env vars and bluetoothctl, is that what you are thinking?

AlexProgrammerDE, 2020-06-05:
Yes. Deleting some lines of code and setting the bluetooth reconnect volume to the mac of the Swivel would get it reconnecting to the Swivel on every boot.
1. Deleting some lines and paring once via terminal
2. Get the mac adress of the Swivel (We can use nano to edit the reconnect entry)

tmigone, 2020-06-05:
For point 1, the fact that you depend on pairing via terminal once is what I don't like too much as it's not very user friendly. But you are right that it could work. If you want to give it a shot when you have free cycles let's do it! Probably hard to test since you won't have the Swivel, but if @Sharpey79 also helps you guys could make it work!

Sharpey79, 2020-06-06:
Yeah, I'd need to buy the swivel first, which I'm happy to do since I can use it with Bluetooth speakers as a backup.

Sharpey79, 2020-06-06:
Also, yes I could just connect the preamp to the pi 3.5 jack... But will balena sound broadcast the signal to other pi units in a multiroom setup? My assumption was that balena sound was only configured to do audio out via the 3.5jack.

Sharpey79, 2020-06-06:
Also, thanks guys. Really appreciate your help.

tmigone, 2020-07-17:
Update, I bought a bluetooth emitter/receiver and will be trying this out soon.

Lui5Jacint0, 2022-02-03:
> Yeah, I'd need to buy the swivel first, which I'm happy to do since I can use it with Bluetooth speakers as a backup.

Did you get it to work?