iotsound/iotsound #208: Spotify Initial Volume

gtihubuser, 2020-06-05:
### Spotify connect uses the 'SYSTEM_OUTPUT_VOLUME' variable as initial volume.
If this volume is set to 100% it means that when i connect to Spotify-connect, my naeghbours and family members are blown away :). (I am using a raspberry pi 3's with Hifi Berry Amp).

If I set the SYSTEM_OUTPUT_VOLUME to like 25%, the maximum volume for the soundcard is set to 25% but Spotify also drops down to 25%. which means that it now starts at 25% of the 25%. If i turn up the volume to 100% on the spotify connect app, the soundcard will play at 25%.

This volume level is (re)set whenever the Spotify service is restarted.

I have fixed this by adding a new variable called SPOTIFY_INITIAL_VOLUME to the _start.sh_ in the Spotify folder.

This variable is then used when starting the service.
I don't know (yet..) how to make a branch or request of how to implment this, so i'm hoping that a kind soul could help with this.

Here is the code for _start.sh_ in the Spotify folder (balena-sound-master\spotify):

```
#!/usr/bin/env bash
 
#Check for incompatible multi room and client-only setting
if [[ -n $DISABLE_MULTI_ROOM ]] && [[ $CLIENT_ONLY_MULTI_ROOM == "1" ]]; then
  echo “DISABLE_MULTI_ROOM and CLIENT_ONLY_MULTI_ROOM cannot be set simultaneously. Ignoring client-only mode.”
fi
 
#Exit service if client-only mode is enabled 
if [[ -z $DISABLE_MULTI_ROOM ]] && [[ $CLIENT_ONLY_MULTI_ROOM == "1" ]]; then
  exit 0
fi

# Set the device broadcast name for Spotify
if [[ -z "$DEVICE_NAME" ]]; then
  DEVICE_NAME=$(printf "balenaSound Spotify %s" $(hostname | cut -c -4))
fi

# Set the system volume here
SYSTEM_OUTPUT_VOLUME="${SYSTEM_OUTPUT_VOLUME:-75}"

# Set the Initial volume here
SPOTIFY_INITIAL_VOLUME="${SPOTIFY_INITIAL_VOLUME:-10}"

# Set the Spotify username and password
if [[ ! -z "$SPOTIFY_LOGIN" ]] && [[ ! -z "$SPOTIFY_PASSWORD" ]]; then
  SPOTIFY_CREDENTIALS="--username \"$SPOTIFY_LOGIN\" --password \"$SPOTIFY_PASSWORD\""
  printf "%s\n" "Using Spotify login."
fi

# Use pipe backend if multi room is enabled
# Don't pipe for Pi 1 family devices since snapcast-server is disabled by default
if [[ -z $DISABLE_MULTI_ROOM ]] && [[ $BALENA_DEVICE_TYPE != "raspberry-pi" ]]; then
  SPOTIFY_BACKEND="--backend pipe"
  SPOTIFY_DEVICE="--device /var/cache/snapcast/snapfifo"
fi

# Start raspotify
exec /usr/bin/librespot --name "$DEVICE_NAME" $SPOTIFY_BACKEND --bitrate 320 --cache /var/cache/raspotify --enable-volume-normalisation --linear-volume --initial-volume=$SPOTIFY_INITIAL_VOLUME $SPOTIFY_CREDENTIALS $SPOTIFY_DEVICE
```
[image]

AlexProgrammerDE, 2020-06-05:
@gtihubuser you could have achieved this too by setting System output volume for each service via the the device service variable tab. I think we have to  larify for every user that this is a option.

gtihubuser, 2020-06-05:
Oh. I did not know that. I'll play with this right away :)
Thanx @AlexProgrammerDE 

So the SYSTEM_OUTPUT_VOLUME in the _Device Variables_ if for the 'soundcard'?
And the SYSTEM_OUTOUT_VOLUME ind the _Device Service Varibles_ Is for the specific 'Service'. Nice :D
Like this i guess:
[image]

Yeah, it could be a good idea to show or list down the options somehow. ;)

AlexProgrammerDE, 2020-06-05:
Yeah its like that. Device variable sets it for every service and service variables for every service individually. I will make later a PR which adds it to the documentation.

gtihubuser, 2020-06-05:
Great!
I think we can close this one then?

AlexProgrammerDE, 2020-06-05:
I will open a PR and link it with this issue. It closes it automaticly when its merged.

tmigone, 2021-06-15:
This was addressed with the release of 3.0 last year. Volume is now controlled by the audio block, can be edited at runtime visiting `http://balena.local` or at startup via env var `SOUND_VOLUME`

closed 2021-06-15 by tmigone