iotsound/iotsound #209: DeviceSupport.md has poofed

jtagcat, 2020-06-05:
https://sound.balenalabs.io/docs/DeviceSupport.md 404

tmigone, 2020-06-05:
Hey @jtagcat thanks for reporting it. We fixed this but haven't deployed yet. 

The correct URL in case you want to check it is: https://sound.balenalabs.io/docs/supported-devices/

jtagcat, 2020-06-06:
The notes (numbered 1-4) appear as links. They are not. It would be nice, if it could be 'show-on-hover'.

tmigone, 2020-06-15:
Hey thanks again for the suggestion. Show on hover would be nice though it's not possible at the moment since the stack combo (GitHub markdown + landr) we use to generate the pages does not allow for something like that. I will merge a PR soon to at least remove the link aspect those items have. Thanks.

jtagcat, 2020-06-15:
I guess can be closed as wontfix for now.

closed 2020-06-15 by jtagcat