iotsound/iotsound #214: Support for Pirate Audio DAC LCD

tommfy, 2020-06-11:
I've got a Pimoroni Pirate Audio 3W Stereo Amp (https://shop.pimoroni.com/products/pirate-audio-3w-stereo-amp) and have decided to switch over from the provided software (Pirate Audio/Mopidy) to BalenaSound as the Spotify Connect feature is fab for my reconditioned radio project and see much improved boot times. The DAC itself works flawlessly on BalenaSound at the moment thanks to your previous instructions but it would be awesome if could make use of the IPS LCD and onboard buttons. Even if the screen just displays a user-configurable splash screen or displays album artwork.

I know the Pirate Audio line is still pretty new but it would be awesome to see once you've had a chance to play with one!

tmigone, 2020-06-15:
Hey @tommfy glad it worked! I'm curious, did you use the `hifiberry-dac` dtoverlay or a different one? 
About the LCD, yes! I'm planning on getting one of this in the near future, so hopefully we can add support for that!

tommfy, 2020-06-15:
It's a great alternative to the Pirate-Audio Mopidy front-end! That's right, I just used the standard `hifiberry-dac` dtoverlay as mentioned in your docs (https://sound.balenalabs.io/docs/dac-configuration/).

That's great to hear about the LCD, that would be killer - looking forward to seeing it whenever you get your hands on one!

closed 2020-06-15 by tommfy