iotsound/iotsound #215: Add support for Mopidy

shaunmulligan, 2020-06-12:
**Describe the solution you'd like**
It would be great if we could add a [Mopidy](https://mopidy.com/) container that can be used and shared to all the rooms, with the ability to mount a USB drive with audio on it.

ChestersGarage, 2020-09-24:
@shaunmulligan Are you specifically interested in Mopidy?  Or would MPD work for you case as well?  I am playing around with MPD right now, for other reasons.  But I'm not yet clear on the differences between the two, and whether they are interchangeable.  I did see this morning that one of the MPD clients I tried on Android said it is not compatible with Mopidy.  From what I've read, Mopidy looks like an off-shoot of MPD.

scls19fr, 2020-11-24:
Hello,

I'm also looking for a way to connect RadioDroid to BalenaSound (or boombeastic https://github.com/balenalabs/boombeastic/issues/59 )
My use case would be to cast webradios listed on https://www.radio-browser.info/ through RadioDroid Android application to a Raspberry Pi 2 model B (which doesn't have bluetooth support)
https://github.com/segler-alex/RadioDroid/issues/936
RadioDroid seems to have MDP support.

Kind regards

tbelaire, 2021-03-15:
I am not the OP, but Mopidy provides a websockets interface which makes it nice to control from a web UI.  That would be one reason to use a Mopidy implementation over other mpd servers.

tbelaire, 2021-03-20:
To follow up, since you can run Mopidy as either client or server nodes, it would be really nice to share configuration.

https://www.home-assistant.io/blog/2016/02/18/multi-room-audio-with-snapcast/

cristiioan, 2022-01-01:
Any update on this?
I'm looking for the same solution.