iotsound/iotsound #222: Error building

jackstirzaker, 2020-06-18:
I am building from balena CLI on my Windows laptop and trying to put it on a Raspberry Pi 3. During the build, the CLI says:
`[Error]             An error occured: (HTTP code 404) no such image - no such image: 94d74402d54b: No such image: 94d74402d54b:latest `

AlexProgrammerDE, 2020-06-18:
I had the same issue some time ago. You might have resolved it by trying it again. @tmigone do you have a idea why this occurs sometimes? Is this a issue with balena builders or is it a client side thing?

jackstirzaker, 2020-06-18:
I am currently trying to build it again.

jackstirzaker, 2020-06-18:
The build has succeeded.

tmigone, 2020-06-19:
Hey @jackstirzaker @AlexProgrammerDE, this is indeed a temporary issue with our builders. Retrying the build will solve the problem as you suggested.

closed 2020-06-19 by tmigone