iotsound/iotsound #224: spotify not working

RJStevenson, 2020-06-22:
I installed balena on my raspberry pi zeri w.
I got the Bluetooth to work by going in the /usr/share/alsa/alsa.conf
and changing the defualts.ctl.card and defaults.pcm.card
value to match that of my USB to 3.5mm jack.
but the same fix is not working for Spotify. I can see the device in Spotify and can connect to it but unable to play audio. as soon as I play audio it disconnects

RJStevenson, 2020-06-22:
these are the error logs in the terminal

22.06.20 15:10:53 (+0200)  spotify  [2020-06-22T13:10:53Z INFO  librespot_playback::player] <Doin' it Right (feat. Panda Bear)> loaded
22.06.20 15:10:54 (+0200)  snapcast-client  2020-06-22 13-10-54 [Error] (worker) Exception in Controller::worker(): connect: Connection refused
22.06.20 15:10:54 (+0200)  snapcast-client  2020-06-22 13-10-54 [Error] (stop) Error in socket shutdown: Transport endpoint is not connected

jtonello, 2020-07-15:
I have a user who is unable to get Spotify Connect to work and I ran my own version an RPi 3 (not 64). The spotify service will not start and shows an exited status. The balenaSound_251e device shows up for a moment, then disappears. I've confirmed this on my mobile Spotify app, Web browser and installed desktop app.

tmigone, 2020-07-17:
Hi @RJStevenson, unfortunately there is no easy fix for the problem you are experiencing. The fix you applied to use the USB soundcard will be a hit or miss in my experience. This will be fixed however in the next major version of balenaSound.

Also @support-agents i think the linked JF thread does not relate to the original issue.

tmigone, 2020-09-21:
Hey @RJStevenson, with the release of balenaSound v3.0 USB soundcards should be supported out of the box. Let us know if you have trouble setting it up or have any other questions. Thanks!

closed 2020-09-21 by tmigone