iotsound/iotsound #225: Multiroom adjust volume on each device

fbacker, 2020-06-22:
**Is your feature request related to a problem? Please describe.**
Multiroom with Spotify I need one device to have volume 30 and another 80 for them to match. This because the connected output on each device is different. DAC, HDMI or 3.5mm jack.

**Describe the solution you'd like**
Set custom audio variable on each device that will 'cap' the volume. e.g. ```SYSTEM_LIMIT_VOLUME 30```. 

When device getting volume of:

* 100% it will be 30%. 
* 50% volume will be 15%. 
* 25% will be 7.5%

**Describe alternatives you've considered**
Might already exist a solution for this but can't find/understand it.

**Additional context**
Raspberry Pi 3
balenaOS 2.51.1+rev1

frederickjansen, 2020-06-24:
Duplicate of #197

tmigone, 2020-07-17:
Yeah this is a duplicate of that linked issue. I'll close, let's use the other one.

closed 2020-07-17 by tmigone