iotsound/iotsound #228: Amazon Echo/Alexa Integration

lundbergj, 2020-06-25:
Huge fan of the project. I've seen others pose the question, but there doesn't seem to be a clear answer. Can balena-sound be used by Amazon Echo/Alexa devices to leverage multi-room? I've seen suggestions that by adding a docker for Alexa it could be integrated, but that seems like mostly in theory. Curious if there's been any thought given to allowing it to function like a wifi-speaker for integration to other services. Don't get me wrong, the current solution is great, just curious if there are plans to take it further.

tmigone, 2020-06-25:
Hi @lundbergj, project maintainer here. 
We haven't done any real assessment or tests as to wether it's feasible to have this integration. I've seen one or two times people asking for this in our forums, but no significant progress really. I know `AlexaPi` is no longer under active development, but i've seen a few other alternatives. It's definitely something that we would like to add to the project eventually but it's not really the focus at the moment. The next major release of balenaSound will make adding new `plugins` a lot more easier, so I don't think we'll tackle this until then.

jolu1705, 2021-03-05:
> 
> 
>  but i've seen a few other alternatives.

Can you put in some links to the alternatives you mentioned here please?

pschaub, 2022-08-05:
It would be awesome if Alexa would be supported. It is currently not possible to play music on Spotify on different devices at the same time (e.g. on your own speakers in one room and on Alexa devices in another room).