iotsound/iotsound #230: deploy failed for snapcast-client and snapcast-server

ejannett, 2020-07-02:
Hi,
Cannot deploy balenaSound due to build failure  on snapcast-server and snapcast-client

**To Reproduce**
1. git clone https://github.com/balenalabs/balena-sound.git
2. balena push _appname_
3. error raised as wget is failing on wrong URL
```
[snapcast-client]   The command '/bin/sh -c wget 'https://github.com/badaix/snapcast/releases/download/v'$SNAPCAST_VERSION'/snapclient_'$SNAPCAST_VERSION'-1_armhf.deb'' returned a non-zero code: 8
[snapcast-server]   The command '/bin/sh -c wget 'https://github.com/badaix/snapcast/releases/download/v'$SNAPCAST_VERSION'/snapserver_'$SNAPCAST_VERSION'-1_armhf.deb'' returned a non-zero code: 8
[Info]              Uploading images
[Success]           Successfully uploaded images
[Error]             Some services failed to build:
[Error]               Service: snapcast-server
[Error]                 Error: The command '/bin/sh -c wget 'https://github.com/badaix/snapcast/releases/download/v'$SNAPCAST_VERSION'/snapserver_'$SNAPCAST_VERSION'-1_armhf.deb'' returned a non-zero code: 8
[Error]               Service: snapcast-client
[Error]                 Error: The command '/bin/sh -c wget 'https://github.com/badaix/snapcast/releases/download/v'$SNAPCAST_VERSION'/snapclient_'$SNAPCAST_VERSION'-1_armhf.deb'' returned a non-zero code: 8
[Info]              Built on arm01
[Error]             Not deploying release.
```

**Device information**
 - Board (hardware info): Raspberry zerow
 - Device type (as seen in balenaCloud): Raspberry Pi (v1 / Zero / Zero W)
 - balenaOS version: balenaOS 2.48.0+rev1
 -balena cli version : 12.3.6
 
**Additional context**
It seems that the arguments defined for snapcast-client and snapcast-server are not taken into account by docker-compose 
Once the version is  put within snapcast-server/Dockerfile.aarch64 and snapcast-client/Dockerfile.aarch64. Everything is fine
i.e
```
diff --git a/snapcast-client/Dockerfile.aarch64 b/snapcast-client/Dockerfile.aarch64
index 4520c9e..cf2fb97 100644
--- a/snapcast-client/Dockerfile.aarch64
+++ b/snapcast-client/Dockerfile.aarch64
@@ -15,8 +15,8 @@ RUN install_packages \
   libvorbis0a \
   wget
 
-RUN wget 'https://github.com/badaix/snapcast/releases/download/v'$SNAPCAST_VERSION'/snapclient_'$SNAPCAST_VERSION'-1_armhf.deb'
-RUN dpkg -i 'snapclient_'$SNAPCAST_VERSION'-1_armhf.deb'
+RUN wget 'https://github.com/badaix/snapcast/releases/download/v0.19.0/snapclient_0.19.0-1_armhf.deb'
+RUN dpkg -i 'snapclient_0.19.0-1_armhf.deb'
```

tmigone, 2020-07-10:
Hi @ejannett, I just tried replicating this error but could not do so. Indeed it looks like it's not picking up the compose arg values. 
Just to confirm, you tried to push the app using `balena push` not `git push` correct? If so, what version of the balena CLI are you using (`balena -v`)?

tmigone, 2020-07-10:
Just saw you posted the CLI version above.

tmigone, 2020-07-10:
Also, what's the application arch (not the device type), is that `rpi`? You can check it on the apps page in the dashboard.

ejannett, 2020-07-10:
Hi,
this is rasberryPi zeroW

ejannett, 2020-07-10:
this is indeed "rpi"

tmigone, 2020-07-10:
Thanks,  last question, you are working with our production cloud right (https://dashboard.balena-cloud.com/apps)?

ejannett, 2020-07-10:
> Thanks, last question, you are working with our production cloud right (https://dashboard.balena-cloud.com/apps)?

yes

tmigone, 2020-07-14:
hey @ejannett, this was identified as an issue with our image builders. It has been solved now so i'll close the issue.

closed 2020-07-14 by tmigone