iotsound/iotsound #232: Services not running

captain-raccoon, 2020-07-06:
Hello,

I was able to flash the SD-Card and get my Pi on with the belena cloud, but when I look at the pi I do not see any of the services running as specified in the setup guide. 

I followed the guide up until that point and I am stuck.  I provisioned both as a production and development edition.  Version number was balenaOS 2.48.0+rev1.  I am using a Raspberry Pi 2 Model B.  I do know that the Pi 2 has not been fully tested but following the guide, I do not believe I am using Multi-Room as I have added the environment variable to disable it.  

Any advice would be appreciated.  Also let me know if you need anymore info.

Thanks in advance!

AlexProgrammerDE, 2020-07-07:
@captain-raccoon it seems like you didn't push the code to balena cloud.

captain-raccoon, 2020-07-07:
Ah I see.  I was just following the guide and I didn't see that as a step.  But this does make sense.  I was under the assumption that when you flashed the SD Card that the OS contained everything.  Thank you for pointing me in the right direction!

closed 2020-07-07 by captain-raccoon