iotsound/iotsound #233: way to get "now playing" data for other projects

Iamnihc, 2020-07-10:
Although many people have suggested adding a gui to balenaSound, I have a different idea for an improvement that could lead to this. have some (documented) way to get information about what the device is doing at the moment, and possibly a way to change it. Some useful information would be:
Connection type (airplay/spotify/bluetooth)
Song/track name (ex: Never gonna give you up)
Song artist (ex: rick Astley)
Current volume (ex 75%)

More suggestions
being able to change current volume
next/previous/play pause
album art

another option is to add this to the Bluetooth connection scripts, but allow it for all connection types with user accessible data. 

I personally would like to use this in conjunction with balenaDash to make it possible to see what is playing on speakers.

tmigone, 2020-07-17:
Hi @Iamnihc , this is definitely coming in the next major release of balenaSound. Maybe not all at once, but we plan on having event hooks and an API to be able to interact programatically with this information.

BestRazer, 2021-09-02:
Is this still coming?

TvdBrink, 2022-08-16:
@tmigone Any update on this one?

Koophear, 2024-01-30:
Any updates on this? @tmigone