iotsound/iotsound #234: Nfc bluetooth pairing

AlexProgrammerDE, 2020-07-11:
**Is your feature request related to a problem? Please describe.**
I saw devices which can pair with my device by simply using nfc pairing. I will check it out later and try to get it to balenaSound.

**Describe the solution you'd like**
https://www.taterli.com/nrf5/nrf5/nfc_ble_pair_lib_dox.html

AlexProgrammerDE, 2020-09-16:
Found this: `Bluetooth Secure Simple Pairing: connect to a Bluetooth device, contains information about the target device like the service class (C# only)`
Here: https://github.com/andijakl/ndef-nfc
Its C# only so imma keep searching, but at least i have a name for that stuff now.

AlexProgrammerDE, 2020-09-16:
Found a python package which is capable of pairing: https://ndeflib.readthedocs.io/en/stable/records/bluetooth.html

tmigone, 2020-09-21:
Nice @AlexProgrammerDE! That last one looks promising. Can you re-open this in the bluetooth block repo? Since we now have that we should move it there.

closed 2020-09-21 by tmigone