iotsound/iotsound #241: Plugin: Spotify

tmigone, 2020-07-17:
Rewrite image to use pulseaudio backend if possible. librespot has a pulseaudio backend (https://github.com/librespot-org/librespot#additional-dependencies) but raspotify does not expose it i think. Need to investigate

AlexProgrammerDE, 2020-07-18:
I will take a look at it. In the case it doesn't support it i will directly build librespot. Btw i have now spotify premium, so i will test i now too.

AlexProgrammerDE, 2020-07-18:
Ah and btw i think raspotify just installs librespot on the PI and we are directly executing it. Maybe building it ourselfes is better? Cuz that project might get outdated.

AlexProgrammerDE, 2020-07-18:
@tmigone i made a project with the stuff. Just needs some testing. We can make a template repo for balenaSound plugins so developing is faster.

AlexProgrammerDE, 2020-07-19:
If the spotify project work will i create the template and share it here.

AlexProgrammerDE, 2020-07-28:
@tmigone i got librespot working. Everything seems to work fine, but i get no audio output. I will keep it like this. Because i dont know if its spotify or the audio primitive causing it. Can you please take a look at it?

tmigone, 2020-07-28:
hey @AlexProgrammerDE I'm having a hard time compiling the rust packages with livepush. Did you manage to push it using livepush? Maybe it's too much but if I deploy it works? Going to try that later

AlexProgrammerDE, 2020-07-28:
hi @tmigone i am not entirely sure what you mean with "livepush" Im just pushing the code to my device through the `blena push appname` command. Building the librespot rust package works fine for me. Im using a raspberry pi 4. What device are you pushing to?

AlexProgrammerDE, 2020-07-28:
Ah and i forgot to mention that raspotify as you said only compiles with alsa support. Thats one more reason to compile it ourselves.

tmigone, 2020-07-28:
I was pushing to a Pi4 using livepush (`balena push <IP of device in local mode>`), if you haven't I recommend you check it out because it's awesome and super fast: https://www.balena.io/blog/living-on-the-edge-with-livepush-and-other-cli-improvements/

Except in this case is not good, so maybe try it for some other project 😆 
Livepush basically builds the docker-compose file on the device (so you don't need to push and create a release everytime), and it seems that librespot takes an insane amount of time to build on a pi, so yeah not a great idea hahah. 

I'm trying on my computer now

tmigone, 2020-07-29:
@AlexProgrammerDE the plugin works great! There are a few missing bits for the audio block i haven't pushed yet and that's why you don't hear audio, but I've tested locally and it's 👍 
I'll try upload a few diagrams to the google doc I shared with you that hopefully will make understanding the audio block easier. 

Also, @chrisys didn't like the name `primitives` 🤣 🤣 🤣  so they are now called `blocks` 🤭

tmigone, 2020-08-05:
Just pushed a few changes that fix the audio block, spotify is now working 💪 
I couldn't get it to work for pi zero / pi 1 though. It throws a weird error, did you see the same?

tmigone, 2020-08-05:
The other problem is that the container is massive (1GB). I looked into cross-compiling the binaries (similar to what raspotify does). I managed to get it done but not for pi zero/1. So won't push those changes yet.

AlexProgrammerDE, 2020-08-05:
@tmigone hi. I couldn't get the chance yet to test it on older devices. I waited for the sound fixing update 😉 i will test it out soon. What does cross compiling do exactly?

tmigone, 2020-08-05:
So the idea is that we have the binary, the librespot program, already compiled and stored in a folder. Then the dockerfile only needs to copy that file and run it, no need to get rust, install cargo and build it from source, which is the part that takes a lot of time. 
These binaries are different for each architecture (aarch64, armv7, x86, etc) and to get them you need to compile on a device that runs the target architecture. Cross-compilation allows you to build binaries for different architectures from a single development machine (which runs on 1 specific architecture). So from your laptop you can build for aarch64 (raspberry pi 4) for example. Problem is that it's not very easy to setup and get it to work. I managed to do it for most archs, but for armv6 (pi zero, pi 1) it's very hard and I'm not really great with rust either.

tmigone, 2020-08-05:
Raspotify for example uses [this](https://github.com/dtcooper/raspotify/blob/master/Dockerfile#L29) which works for all raspberry pi architectures. But I can't get it to build the pulseaudio backend.

tmigone, 2020-08-05:
I'll clean up my dockerfiles for the other archs and upload so you can see.

tmigone, 2020-08-12:
TODO

- [ ] Not working for Pi Zero/Pi 1
- [ ] Not working for Pi 4 / Pi 3 when using lesser app archs (pi0/1)

tmigone, 2020-08-24:
So update on this. After spending WAY more time than expected I could not get a binary that worked fine with the `pulseaudio` backend. A very TL;DR of my experience with this:

- Installing with cargo or building on device --> took too long and was not supported for Pi Zero.
- Pre-building the binaries using cross-compilation --> a pain to setup, though it almost worked. I could not get a binary for Pi Zero, and generally the binaries didn't work if they weren't running on a different base dockerfile (example: Pi 4 binary did not work on a Pi 4 if it ran on a Pi 0 app).

So I resorted to keep using `raspotify` with the ALSA backend but using an ALSA bridge as described here: https://github.com/balenablocks/audio#sendreceive-audio

From conversation with the librespot maintainer they seem to be wanting to host pre-built binaries with pulse support but it's not high up their priority list. If it happens we should revisit this.

closed 2020-08-24 by tmigone