iotsound/iotsound #242: Plugin: Airplay

tmigone, 2020-07-17:
Convert image to use pulseaudio backend if possible. According to docs [here](https://github.com/mikebrady/shairport-sync#what-else) it has pulseaudio backend, though it might not be included on the apt-get package, so maybe we need to build from source?

minix1234, 2020-07-19:
Mr. Brady has done some significant work creating a common docker image for shairport-sync. please see docker file at https://hub.docker.com/r/mikebrady/shairport-sync/dockerfile

minix1234, 2020-07-19:
ps. the Debian package is very much outdated

AlexProgrammerDE, 2020-07-19:
@minix1234 what do you mean with "outdated"? Does the team no longer maintain the project or are they just not publishing the code? @tmigone if thats the case we should maybe build it ourselfes.

minix1234, 2020-07-19:
the Debian package repository is at revision 3.2.2 (October 2018) while current GitHub release is 3.3.7.  I believe Chris boot was maintaining the Debian package/git repository.  it has since been stale.

minix1234, 2020-07-19:
and just for clarification I'm not sure why that's the case.  I see Mr boot has been doing work on the Debian version of the repository in late 2019.  but the packaged version of the application is still 3.2.2.

tmigone, 2020-08-24:
@minix1234 do you know if the the docker image works across the entire raspberry pi family of devices?

minix1234, 2020-08-26:
Evening,

It should yes.  I have a version that is working very well on RP zero.  so I assume it works with all the rest.

check my forked Belena-Sound repository's Airplay folder.  the docker image is generic but I have modified the start script to suit my needs.

tmigone, 2020-08-26:
Hey, thanks for the information. I'll take a look but this sounds like a great idea. Thanks for bringing it up!

AlexProgrammerDE, 2020-12-29:
Seems all done here

closed 2020-12-29 by AlexProgrammerDE