iotsound/iotsound #248: Add support for BOSS I2S DAC V 1.2

fa0lan, 2020-07-18:
Hey guys,
I really enjoy using balena-sound. I've got a DAC (BOSS I2S DAC V 1.2) which is not officially supported in your `/docs/04-dacs.md`. I have tried adding the custom configuration variable `BALENA_HOST_CONFIG_dtoverlay` with the value of `allo-boss-dac-pcm512x-audio`. No success so far. I have also modified the DT parameters as explained in your [guide](https://www.balena.io/blog/turn-your-old-speakers-or-hi-fi-into-bluetooth-receivers-using-only-a-raspberry-pi/?utm_medium=etcher&utm_source=efp&utm_campaign=balena-sound&utm_content=v6#setuptheraspberrypi).
The `snapcast-client` throws tons of errors and i tried spotify connect, airplay and bluetooth playback. Non worked for me.
I previously used the DAC in combination with the OS Volumio which always worked fine with this DAC.

tmigone, 2020-09-03:
Hey @fa0lan, as per [this](https://allo.com/documents/sparky/tech-manuals/BOSS-DAC-Tech-Manual.pdf) guide, you are using the correct dtoverlay. My guess is that the DAC is working but those `snapcast-client` errors don't sound good. If you can share them we can figure out what is going on.

tmigone, 2020-09-22:
Closing due to inactivity. Please reopen if still having problems.

closed 2020-09-22 by tmigone