iotsound/iotsound #254: no sound after install hifiberry digi+

stnorton, 2020-07-22:
I've installed the hifiberry digi+ and followed the configuration instructions in the docs, i.e. audio=off and BALENA_HOST_CONFIG_dtoverlay=hifiberry-digi

Since installing it, the only output I get is a low buzzing noise. When I connect via Spotify connect or bluetooth, it will show as connected and playing music, but I get no audio via the speakers. The Pi is plugged into an amp (an old-ish Pioneer one), and other things connected to the amp work normally, and I've tried other inputs on the amp as well. In addition, I've tried a different cable, checked that the DAC is connected correctly, and rebooted the device multiple times with no luck.

This seems to happen to people (based on googling) when they have the wrong overlay set, but I've double-checked that multiple times. 

Happy to provide more info - I've reached about the end of my debugging skills and I'm not very familiar with DACs. This was working before installing the DAC, so I would hazard a guess that the issue lies there somewhere.

**Device information**
 - Board (hardware info): Raspberry Pi 4 Model
 - Device type (as seen in balenaCloud): Raspberry Pi 4
 - balenaOS version: balenaOS 2.51.1+rev1

tmigone, 2020-07-23:
Hey @stnorton have you tried the DAC with another application or OS? I only ask because we've helped people troubleshoot DACs in the past and many times it turns out the DAC was faulty 😁

tmigone, 2020-09-03:
Closing due to inactivity. Feel free to reopen if you are still experiencing problems. Cheers.

closed 2020-09-03 by tmigone