iotsound/iotsound #255: Deploy failure - snapcast bad signature

middric, 2020-07-23:
**Describe the problem**
Pushing a release of master or 2.4.8 and I'm seeing snapcast-server and snapcast-client errors resulting in a deployment failure:

```sh
[Error]             Some services failed to build:
[Error]               Service: snapcast-server
[Error]                 Error: The command '/bin/sh -c install_packages snapcast-server=$SNAPCAST_VERSION --repository=http://dl-cdn.alpinelinux.org/alpine/edge/community' returned a non-zero code: 1
[Error]               Service: snapcast-client
[Error]                 Error: The command '/bin/sh -c install_packages snapcast-client=$SNAPCAST_VERSION --repository=http://dl-cdn.alpinelinux.org/alpine/edge/community' returned a non-zero code: 1
[Info]              Built on arm03
```

and

```
[snapcast-server]   (1/1) Installing snapcast-server (0.20.0-r0)
[snapcast-server]   ERROR: snapcast-server-0.20.0-r0: BAD signature
[snapcast-server]   
[snapcast-server]   1 error; 73 MiB in 83 packages
[snapcast-client]   (1/1) Installing snapcast-client (0.20.0-r0)
[snapcast-client]   ERROR: snapcast-client-0.20.0-r0: BAD signature
```

**To Reproduce**
Steps to reproduce the behavior:
1. git clone https://github.com/balenalabs/balena-sound.git
2. git checkout v2.4.8 (master has same issue)
3. balena push appname

**Device information**
 - Board (hardware info): Raspberry Pi Zero W
 - Device type (as seen in balenaCloud): Raspberry Pi (v1 / Zero / Zero W)
 - balenaOS version: balenaOS 2.48.0+rev1, balena-cli 12.9.1

middric, 2020-07-23:
Following up that setting SNAPCAST_VERSION to 0.19.0-r1 and switching from edge to the latest-stable branch of the alpine repo results in a successful deploy. I can see that the snapcast apk's in the edge branch were modified on 22-Jul-2020 16:09 so presumably they are invalid and dont match the expected signature

tmigone, 2020-07-23:
hey @middric thanks for bringing this up. Looks like the best alternative for the moment is to rely on the latest available version without resorting to edge repositories. This [PR](https://github.com/balenalabs/balena-sound/pull/256) should fix the problem. I bumped the alpine version to 3.12 so that means snapcast 0.19.

tmigone, 2020-07-23:
Should be fixed by v2.4.9, thanks again!

closed 2020-07-23 by tmigone

middric, 2020-07-23:
:+1: Thanks @tmigone