iotsound/iotsound #259: fleet-supervisor cycles service online/offline

netzwerg-au, 2020-07-29:
I have a multi-room deployment with 2 x RPi 4 and 1 x RPi 3B+ all connected via wifi.  I noticed intermittent cutouts (every few minutes) on one of the RPi 4s, regardless of whether I am connecting directly to it or to the other RPi4.  I just checked the terminal of the offending RPi 4 and found that the fleet-supervisor service keeps cycling between online and offline.

```
29.07.20 17:26:31 (+1000)  fleet-supervisor  Fleet subscriber > service.offline Fleet publisher#3eca85f8-cf26-4f0e-ae46-4fa5c7f7b0e7 on 8000
29.07.20 17:26:31 (+1000)  fleet-supervisor  Fleet publisher > service.offline Fleet publisher#3eca85f8-cf26-4f0e-ae46-4fa5c7f7b0e7 on 8000
29.07.20 17:26:35 (+1000)  fleet-supervisor  Fleet subscriber > service.online Fleet publisher#3eca85f8-cf26-4f0e-ae46-4fa5c7f7b0e7 on 8000
29.07.20 17:26:35 (+1000)  fleet-supervisor  Fleet publisher > service.online Fleet publisher#3eca85f8-cf26-4f0e-ae46-4fa5c7f7b0e7 on 8000
29.07.20 17:26:47 (+1000)  fleet-supervisor  Fleet subscriber > service.offline Fleet subscriber#05c49cf4-dcc6-4e1d-8a68-f35aba4e0345
29.07.20 17:26:47 (+1000)  fleet-supervisor  Fleet publisher > service.offline Fleet subscriber#05c49cf4-dcc6-4e1d-8a68-f35aba4e0345
29.07.20 17:26:48 (+1000)  fleet-supervisor  Fleet subscriber > service.online Fleet subscriber#05c49cf4-dcc6-4e1d-8a68-f35aba4e0345
29.07.20 17:26:48 (+1000)  fleet-supervisor  Fleet publisher > service.online Fleet subscriber#05c49cf4-dcc6-4e1d-8a68-f35aba4e0345
29.07.20 17:27:11 (+1000)  fleet-supervisor  Fleet subscriber > service.offline Fleet subscriber#05c49cf4-dcc6-4e1d-8a68-f35aba4e0345
29.07.20 17:27:11 (+1000)  fleet-supervisor  Fleet publisher > service.offline Fleet subscriber#05c49cf4-dcc6-4e1d-8a68-f35aba4e0345
29.07.20 17:27:12 (+1000)  fleet-supervisor  Fleet subscriber > service.online Fleet subscriber#05c49cf4-dcc6-4e1d-8a68-f35aba4e0345
29.07.20 17:27:12 (+1000)  fleet-supervisor  Fleet publisher > service.online Fleet subscriber#05c49cf4-dcc6-4e1d-8a68-f35aba4e0345
29.07.20 17:27:15 (+1000)  fleet-supervisor  Fleet subscriber > service.offline Fleet publisher#671d8f02-caa5-4768-b462-e6ef08e79686 on 8000
29.07.20 17:27:15 (+1000)  fleet-supervisor  Fleet publisher > service.offline Fleet publisher#671d8f02-caa5-4768-b462-e6ef08e79686 on 8000
29.07.20 17:27:16 (+1000)  fleet-supervisor  Fleet subscriber > service.online Fleet publisher#671d8f02-caa5-4768-b462-e6ef08e79686 on 8000
29.07.20 17:27:16 (+1000)  fleet-supervisor  Fleet publisher > service.online Fleet publisher#671d8f02-caa5-4768-b462-e6ef08e79686 on 8000
```

I have tried rebooting the offending device, but the issue keeps coming back.

I know that this issue was opened previously, but there was no resolution before it was closed.  Anyone have any ideas what is going on here?  I don't see the same cycling issue on my other RPis.

Rgds,
Brett.

tmigone, 2020-08-10:
Hi @netzwerg-au , can you check if the fleet supervisor service is being restarted or is it just the publisher/subscribers? You can verify this by checking the dashboard and checking the `fleet-supervisor` service logs.

tmigone, 2020-09-03:
Closing this due to inactivity. Feel free to reopen if you are still experiencing problems.