iotsound/iotsound #260: HomeAssistant integration

tmigone, 2020-07-30:
Consider adding homeassistant integration.

DenielTorres, 2020-11-12:
Hi @tmigone! Any update on this enhancement? I'm about to install a couple of in-wall speakers around the house with balena sound and would be awesome to somehow integrate these into HA. Maybe as media players? 

Thanks in advance! 
Daniel

drew-kun, 2021-02-11:
Yeah, would be great! I came across this request hoping to find any way to integrate it into my HA.

FlyersPh9, 2021-02-27:
Also would like to integrate my Balena Sound into Home Assistant.

robsloetjes, 2021-04-07:
Balena Sounds apparently uses Snapcast to enable the multi room audio. Snapcast can be integrated in Home Assistant ([https://www.home-assistant.io/integrations/snapcast/](https://www.home-assistant.io/integrations/snapcast/)).

In my setup I created a host device with Balena Sound. On every client device I installed Snapclients ([https://github.com/badaix/snapcast](https://github.com/badaix/snapcast)), but Balena Sound clients should also work. This should give you a working multi room audio system.

Each device (host and/or clients) can then be added in the configuration.yaml of Home Assistant:

```
media_player:
  - platform: snapcast
    host: 192.168.2.6
```

After a restart, Snapclient devices should appear in your entities list. The devices can then be used in a lovelace dashboard or Node Red for further integration.

I hope this is (a part of) the functionality you are looking for!

Sn1cket, 2024-09-16:
I really would like to see a Home Assistant integration because my Balena Sound instance is running in standalone mode so the Snapcast integration isn't an option.

My main use case would be to have the player status in Home Assistant to use it for some automations.

eataiwo, 2024-09-20:
@Sn1cket I am not an expert, and I stopped using BalenaSound about a year ago, but looking quickly through the BalenaSound documentation, it doesn't seem as if there is a simple way to get the player's status. Maybe if you are able to read the logs from the sound supervisor or audioblock service, but then you would need a way to send that to HA, and as far as I know BalenaSound doesn't have an API so you would have to set that up after the fact with some custom script on your device,  alternatively it's possible to modify the relevant docker files for BalenaSound and deploy that, however again I am not an expert so take what I say with a pinch of salt. Depending on your usecase, you can do a workaround. For example, I mainly only used Spotify, so I ran automation using the Spotify integration in HA to turn on my speakers if Spotify was playing via my BalenaSound device, which Spotify could see. 

All in all, I do not see a HA integration happening anytime soon unless someone is willing to put together some HACS integration

Sn1cket, 2024-09-20:
Yeah, I already had a similar assumption and I looked for alternatives. So far I think I will probably move to moode audio player. It seems like it supports everything I need (BT Audio + API). Their API can be used to integrate into Home Assistant with the RESTful and MPD integration.

eataiwo, 2024-09-20:
@Sn1cket Yes, I moved to Moode Audio as well, and I was pretty happy with it. I haven't done any "direct" integration with Moode and HA yet. I knew it was possible but not plug and play, so I haven't had the time to figure it out. I still just use the Spotify integration, but I wish you luck.