iotsound/iotsound #261: Spotify Song Cache?

Paytah232, 2020-07-31:
Hi, 

I'm wondering if the librespot library used has the functionality to be able to cache songs it has played for future playback. I run my RPis using a 16GB SD (or could go bigger) as sometimes I have intermittent internet dropouts. Most of the time this is okay, but it would be nice if some songs could still play when the internet goes down (without reverting to bluetooth - I also have spotify premium). Actually, that raises another question from me; do these devices work if they're connected to a local network with no WAN access? Could I run them on their own network that allow's them only access to the internet for updates etc (this would also assume bluetooth streaming or possible offline spotify streaming).

If this isn't already a thing, is there a possibility to add it in the future?

As another side note, I tried using the SPOTIFY_LOGIN and SPOTIFY_PASSWORD fleet wide parameters but librespot spat out bad authentication error for me; I know the pass is correct because I copied it from my password manager; is there a ficx for this too?

Cheers!

AlexProgrammerDE, 2020-08-05:
We are already caching songs. Hiw often do you play the songs? It might take a while till librespot decides to make the cache bigger.

tmigone, 2020-08-10:
As Alex commented, we are already using the `cache` option from `librespot`. This should be good enough for internet dropouts though for more prolonged outages it won't cut it. Offline playback would be nice to have but unfortunately it's not something that can be added as it constitutes a violation of Spotify's end user agreement, see [this](https://github.com/librespot-org/librespot/issues/152) issue for more details.

About local networks, yes. balenaSound only requires internet for getting OS/app updates and for using Spotify Connect. All other services should run fine on an isolated network.

closed 2020-09-03 by tmigone