iotsound/iotsound #264: Add support for Waveshare WM8960 DAC HAT

AnomahR, 2020-08-06:
I would like to use the Waveshare WM8960 Audio HAT for the Raspberry Pi Zero W running BalenaSound and would love to see it appearing on the [list of supported DACs](https://sound.balenalabs.io/docs/dac-configuration/)!

I have taken a look at the [installation manual](https://www.waveshare.com/w/upload/5/54/WM8960_Audio_HAT_User_Manual_EN.pdf) but do not know how to install the drivers on the Balena OS.

Any support regarding the matter would be greatly appreciated!

AlexProgrammerDE, 2020-08-06:
@AnomahR after taking a look at the installation programm i found this: `wm8960-soundcard` can you please set this as your dtoverlay? Please let me know if it works

AnomahR, 2020-08-06:
Thanks for your quick response! After setting the `BALENA_HOST_CONFIG_dtoverlay` parameter to `wm8960-soundcard`, I got the following error messages: [dtoverlay_wm8960-soundcard.txt](https://github.com/balenalabs/balena-sound/files/5033548/dtoverlay_wm8960-soundcard.txt)

Also I tried to connect via bluetooth and Spotify Connect but neither way I could hear any sound. The usual "bluetooth-connecting" sound was not played aswell.
 
Hope that helps!

cryptoAlgorithm, 2020-08-06:
I hope that this can be fully working soon as I also have this HAT lying around somewhere and would like to try it out for lower noise and higher volume.

AnomahR, 2020-08-08:
Three days of not doing what I was intended to later I finally made it work! Not the most elegant solution but it works for my Pi Zero W with the WM8960 DAC HAT. In case it might be helpful here is how I did it:

1. Added the `wm8960-soundcard.dtbo` to the overlays folder of the image before flashing it to the SD card (see [instructions](https://forums.balena.io/t/preload-device-tree-overlays/5495/8). Furthermore before flashing, I added the following lines to the `config.txt` file in the boot partition of the image:

```
dtparam=i2s=on
dtoverlay=i2s-mmap
dtoverlay=wm8960-soundcard
```

2. Installed the remaining drivers in a Docker container and added it to my fork of the BalenaSound project. Pushed the current state on GitHub here: https://github.com/AnomahR/balena-sound. The additional container has its own repository: https://github.com/AnomahR/WM8960-Audio-HAT.

3. After deploying the project to my Pi Zero, I changed the DT parameters as suggested in the [DAC configuration](https://sound.balenalabs.io/docs/dac-configuration/): For the DT parameter, `"audio=off"` and then I set the custom configuration variable `BALENA_HOST_CONFIG_dtoverlay` to "wm8960-soundcard".

Thats it! Unfortunately there is no sound when I connect to this specific Pi via Spotify Connect (but it still plays sound if I connect to my other Raspberry using Spotify Connect), maybe you have a clue what could cause the problem? Also not optimal: Now this additional container is installed on all other devices doing nothing... definitely looking forward to the DAC HAT being "officially" supported!

kevincastejon, 2020-08-30:
I'm also highly interested on using that hat, i'm sad to read that the solution above does not allow Spotify Connect cause it's the main usage of my project...

tmigone, 2020-09-03:
hey @AnomahR , from your notes I think you should be able to get the DAC to work following [this](https://sound.balenalabs.io/docs/dac-configuration) instructions but using both dtoverlays as such:

`BALENA_HOST_CONFIG_dtoverlay="i2s-mmap", "wm8960-soundcard"`

It also should not affect the Spotify Connect usage at all, so maybe there is something else going on there.

kevincastejon, 2020-09-05:
@tmigone I just tried that config but still no sound after reboot...

AnomahR, 2020-09-08:
@tmigone You are right about the initial Spotify issues which turned out not to be related to the soundcard. Eventually I got everything to work using my additional container that installs the required kernel module (see repo above), but then realized my soundcard also had a hardware issue, meaning I had to send it back and cannot test your suggested configuration unfortunately. But I will definitely try it once I have the soundcard again!

@kevincastejon Did you modify the config only did you also install the kernel modules following the instructions above?

kevincastejon, 2020-09-08:
@AnomahR Actually I just changed the config, I have to say I don't really know to proceed the steps you mentionned above with docker and stuff... is there a place I can read about ?

AnomahR, 2020-09-10:
To me this looks like the config was correct but there are still driver issues. 

Which hardware are you using? Tested the repo on my Raspberry Pi Zero W with the kernel version 4.19. Waveshare just released new drivers for the kernel version 5.4 but in my repository the older version is still used.

You can check your kernel version when you SSH into the `wm8960-soundcard` Docker container (from the balena dashboard) in the repo by typing

`uname -r`

To test if the soundcard is detected at all, could you give the output of

`aplay -l`?

kevincastejon, 2020-09-10:
@AnomahR 
when I type 'uname -r' I got this : `4.19.118`
when I type `aplay -l` I got this : `bash: aplay: command not found`

shawaj, 2020-12-27:
@kevincastejon you need to add the dt overlay for the card before applying the settings in balenaCloud.

Once you have programmed the SD card you can just add the `wm8960-soundcard.dtbo` file to resin-boot/overlays folder.

By the way, for the `aplay -l` command you would need to run that in a terminal window in the `audio` container.

shawaj, 2021-01-01:
Possible duplicate of #166 as both use the WM8960 chip. Will need testing though

shawaj, 2021-01-02:
Added a PR for the overlay for this to the upstream kernel https://github.com/raspberrypi/linux/pull/4039

@AnomahR @kevincastejon @cryptoAlgorithm - might need your help testing this when it gets pulled in, as I don't currently have one of these cards available

cryptoAlgorithm, 2021-01-02:
Ok sure i have the hat and zero unused so i can installband test it out.

shawaj, 2021-01-02:
@cryptoAlgorithm awesome - will keep you up to date on when the update is pulled in upstream

cryptoAlgorithm, 2021-01-02:
Thanks, always ready to help!

shawaj, 2021-01-07:
good news - this has been pulled into the upstream kernel https://github.com/raspberrypi/linux/pull/4039

will take a bit of time to filter down into here, however it should be possible for you to pull in the overlay manually once it hits the raspberry pi firmware. or you could build the .dtb yourself and pull it in but that is a bit more involved.

shawaj, 2021-01-08:
Interestingly, the [Adafruit Voice Bonnet](https://www.adafruit.com/product/4757) also uses this same wm8960 driver as you can see from this page - https://learn.adafruit.com/adafruit-voice-bonnet/audio-setup

It will be interesting to see if anybody has this so we can test it.

shawaj, 2021-01-13:
@AnomahR @kevincastejon @cryptoAlgorithm this is now in the raspberry pi firmware - https://github.com/raspberrypi/firmware/blob/master/boot/overlays/wm8960-soundcard.dtbo

shawaj, 2021-01-13:
Added the overlay to the PR here - https://github.com/balena-os/balena-raspberrypi/pull/585/files